KiCad/kicad-library #650: STM32F378xx family schematic symbols

mpkopec, 2016-08-19:
Dear Librarians,
I would like to use one of the uCs fromt the mentioned family. They, although, are enormous (see below). Is it really necessary to list every single function of a pin in such situation? The symbol does not even fit on an A4 paper size schematic! Maybe we should introduce some kind of a KLC rule regarding pin descriptions in uCs, maybe dividing uCs into banks, like FPGAs. I'm open to suggestions, and afterwards I could do some changes to this family. What do you think?

[image]

SchrodingersGat, 2016-08-20:
AFAIK @ObKo generated these with a script which pulls out all the pin names, resulting in the long-run-of-text-pin-naming.

A4 is getting rather large! @ObKo would it be possible to clip the names to the primary/secondary/tertiary functions?

poeschlr, 2018-01-20:
A lot of MCU symbols show this problem. This needs to be fixed over in the new repo.
See https://github.com/KiCad/kicad-symbols/issues/187

closed 2018-01-20 by poeschlr