KiCad/kicad-library #655: Why can't make the component rotate 135 degrees?

myyerrol, 2016-08-26:
I found a question about rotating component's reference of value. **Rotating 0 degrees is the same as 180 degrees**. Although I know it is very convenient at some time, I need to different rotation for my kicad project. So how to solve it? Wheather there is a software's configuration or not? If have, please tell me! Thanks a lot!

reportingsjr, 2016-09-04:
I think this is an issue you have with kicad itself rather than the libraries. Feel free to search for/post a bug about this on the official bug tracker: https://bugs.launchpad.net/kicad

If I misunderstood something and this is a bug with the libraries feel free to reopen this.

closed 2016-09-04 by reportingsjr

myyerrol, 2016-09-05:
Ok, thanks a lot!