KiCad/kicad-library #697: Error in libraries syntax when loading with latest nightly build (13 Sep. 2016)

mpkopec, 2016-09-13:
Hello everyone,
I have found that when trying to open eeschema project an error occurs, which seems to be related to the syntax of the library. Below you can find an image with the error message. Did anyone stumble upon this type of error and if so how to deal with it?

[image]

SchrodingersGat, 2016-09-13:
Can you post a copy of your "cmos_ieee.lib" file?

mpkopec, 2016-09-13:
Of course, you can find it below. I have just one more remark - this is freshly cloned from github.

[cmos_ieee.zip](https://github.com/KiCad/kicad-library/files/469892/cmos_ieee.zip)

mpkopec, 2016-09-14:
Ok, the problem does not exist in the stable version, so it's the KiCad itself problem apparently.