KiCad/kicad-library #726: TO-18 model pin 1 tab orientation is incorrect.

stambaughw, 2016-09-27:
I was working on a board and discovered that the pin 1 indicator tab on the TO-18 3D model does not match the footprint silk screen.  According to the documentation that I have seen, it appears that the footprint silkscreen orientation is correct.

[image]

diggit, 2016-10-01:
Thanks, confirmed, model has tab on incorrect position.

If you have some 3D modelling skills, we are open for fixes. Current 3D model looks bad anyway.
Otherwise, I can take a look on it, not sure when.

jkriege2, 2016-10-01:
I made a small fix:
https://github.com/KiCad/kicad-library/pull/734

stambaughw, 2016-10-02:
Look great!  Thank you for the quick response.

Cheers,

Wayne

jkriege2, 2016-10-28:
new 3D models are available now (https://github.com/KiCad/kicad-library/pull/734 merged), footprints still need to be merged (https://github.com/KiCad/TO_SOT_Packages_THT.pretty/pull/12)!

Best,
JAN

SchrodingersGat, 2017-02-23:
Ok to close this one @stambaughw ?

SchrodingersGat, 2017-02-23:
New footprint and 3D model have been merged

jkriege2, 2017-02-26:
This should be fixed now also with the new footprints https://github.com/KiCad/TO_SOT_Packages_THT.pretty/pull/14

closed 2017-02-26 by jkriege2