KiCad/kicad-library #738: Inductor is too big

maxaille, 2016-10-05:
I'm starting using KiCad so sorry if I don't get all the things right away.
The INDUCTOR and INDUCTOR_SMALL are really much bigger than other standard components, where L_Small is the right size compared to R_Small and C_Small, and as I can see there is no L for the normal size present in the "device" library.

Am I missing something or do the Device library lack of consistency on this part ?
[image]

maxaille, 2016-10-05:
I think adding the missing "L" symbol would be nice already, maybe also removing the "INDUCTOR" and "INDUCTOR_SMALL" since they don't seems to respect the actual naming convention. Should I make a pull request with the modifications ?

jkriege2, 2016-10-09:
I addressed both problems in this PR: https://github.com/KiCad/kicad-library/pull/742 Would be OK for you?

JAN

furieuxjk, 2016-10-11:
Please could you also include variable inductor symbol?

jkriege2, 2016-10-11:
THere is a symbol LTRIM (on the right side of the L-symbols-field) ... that enough, or do you need one of those with a third pin + variable?

JAN

furieuxjk, 2016-10-11:
Hmm, I can't find it under device library. I do have CTRIM, for example, but no LTRIM. Usually it is standard inductor symbol, but with arrow over it.

jkriege2, 2016-10-11:
I added it to my PR (https://github.com/KiCad/kicad-library/pull/742), which is not yet part of the device.lib, but you can view my idea here: https://github.com/KiCad/kicad-library/pull/742

Best,
JAN

furieuxjk, 2016-10-12:
Great! Thanks.

closed 2016-11-20 by maximeborges