KiCad/kicad-library #760: No *431 precision adjustable reference in KiCad library?

evanshultz, 2016-10-21:
I don't see [TL431](http://www.ti.com/lit/ds/symlink/tl431.pdf), TL432, TLV431, ZR431, etc. in the KiCad tool or a search of this GH repo. This is a super common part used in numerous power supplies. Probably the majority of them ever made.

I'd love to contribute this part to the library but I'm concerned I'm daft and have missed the part. It's so common, I have trouble believing it's really not there. Can anyone confirm it is, in fact, missing?

jkriege2, 2016-10-21:
Hi!

have a look into the reference.lib ... There are several options of TL431 in different packages ... Possibly you'll have to update the libs from GitHub.

cu
JAN

SchrodingersGat, 2016-10-21:
[image]

@evanshultz looks like there are some matching parts in references.lib

evanshultz, 2016-10-21:
Wow, I really must be losing it. Thank you. I suspected they must be there.

I started by checking out my local PC. I found the library is on my PC but it's not available in the list of libraries, even after manually adding the path. Searching online I found [this](https://forum.kicad.info/t/github-library-workflow/2257/5) and [this](https://forum.kicad.info/t/creating-your-own-offline-libraries-in-new-kicad/1169/15) which helped me get all the libraries in Eeschema. Even after adding the path, I still need to manually add the individual libraries.

So why aren't all the libraries made available by default? And is there no method to push/pull libs based on a GH update or by polling, and then automatically update the local KiCad installation to reflect the library updates? If not automatic, what about a manual method?

This may be a KiCad problem and not a library problem, so I understand if this is the wrong place for this question. If someone can help me better understand library management in KiCad, however, that would be great. Thank you!

closed 2016-10-28 by jkriege2

diggit, 2016-10-28:
@evanshultz 
**symbols**
Because kicad installation carries old libraries (at least stable release), you have to clone current version and manually add its' path to your project. Then you can delete old entries and add all from cloned repository.
Manual method is possible: `git pull` . 
If there is better way, correct me.

**footprint**
You can find current fp-lib-table in this repo. Git variant update automatically.

If there is something unclear, mention my name (`@diggit` ) in message.