KiCad/kicad-library #766: resistor symbol to ANSI/IEEE style

andreymrt, 2016-10-25:
Is it OK to submit a patch to fix the resistor symbol to match ANSI Y32.2/IEEE Std 315?

bobc, 2016-11-04:
That spec provides two symbol shapes for resistor, and KiCad uses one of them.

So *exactly* what change are you proposing, and what is the justification?

andreymrt, 2016-11-04:
The squiggly symbol. The box symbol is mostly used for impedances like transmission lines or generic loads.

Justification: squiggly is far more commonly used and easy to recognize (take a look at any article in JSSC or TCAS)

bobc, 2016-11-10:
The box symbol for a resistor is the preferred international symbol. I understand there are some regions which lag behind international standards...although I don't mind the squiggly myself, it is more distinctive.

Since KiCad recently changed from the sqiggly to the box I guess such a patch would not be accepted.

maximeborges, 2016-11-20:
IEEE and ANSI are americans standard. There is a lot of effort converting all the current libraries to the international IEC standard which, I think, is the best to do. Also in Europe as far as I know you almost never find the squiggly symbol.

andreymrt, 2016-11-22:
Ok, I didn't know that. I always thought that was the standard since it's pretty ubiquitous in nearly all application notes, even from companies like ST and NXP, not to mention textbooks, and lectures from luminaries like Sansen.
I guess I'll just keep my own non-euro libraries then.

closed 2016-11-22 by andreymrt