KiCad/kicad-library #774: Many THT footprints have F.Silks enabled on pads

diggit, 2016-10-27:
As in title, it is problem in many existing FPs. I noticed it in [KiCad/Pin_Headers](https://github.com/KiCad/Pin_Headers.pretty).
This is obviously wrong and should be fixed.

StefanHamminga, 2016-11-03:
I noticed this annoyance too.

A (Linux) quick and dirty way to edit all footprints in a `.pretty` directory, removing the front silkscreen layer from the pads:
```bash
# Find problem pads:
egrep -l '([ \t\s]*\(pad .*[A-Za-z0-9])[ \t\s]*F.SilkS' `find /my/libraries/path/ -iname "*.kicad_mod"`

# in a specific `.pretty` dir:
sed -i -r 's/([ \t\s]*\(pad .*[A-Za-z0-9])[ \t\s]*F.SilkS/\1/' *.kicad_mod

# or if you want to update your entire library:
find /my/libraries/path/ -iname "*.kicad_mod" -exec sed -i -r 's/([ \t\s]*\(pad .*[A-Za-z0-9])[ \t\s]*F.SilkS/\1/' "{}" \;
```

This doesn't account for multi-line pad definitions, but KiCad seems to keep pad definitions in one line so it probably works just fine for all footprints.

pointhi, 2016-11-05:
@StefanHamminga thanks for your bash line. I will create the required PR so we have those useless silkscreen finally gone.

pointhi, 2016-11-05:
Ok, I finished creating all those PR

jkriege2, 2016-11-05:
Great work! I scanned through all of them and merged all PRs (didn't spot any problems and looked into some with KiCAD to check for other problems).

Thanks,
JAN

closed 2016-11-05 by pointhi