KiCad/kicad-library #799: Some libraries have at end of lines the CL+RF symbols

michalkona, 2016-11-20:
These libraries have at end of lines the CL+RF symbols: Worldsemi, device, mechanical, motors, silabs, stm32, switches and texas

It might cause of big conflict.

SchrodingersGat, 2016-12-06:
I see CL+RF at the end of lines of all libraries - can you show an example?

michalkona, 2016-12-08:
Because you are using windows:
https://git-scm.com/book/tr/v2/Customizing-Git-Git-Configuration#Formatting-and-Whitespace
https://help.github.com/articles/dealing-with-line-endings/#platform-windows

bradfa, 2017-02-09:
This is quite widespread actually, it seems to affect .lib, .dcm, and .wrl files.  As of my counting right now (commit 1643c5ced7183a32e251f229ad58ba5849db5629) there are 915 files in the kicad-library repo which have DOS line endings.  This should be able to be fixed by using a .gitattributes file to specify a line ending to always use and making one big cutover commit to change all files in the entire repo at once.

bradfa, 2017-02-09:
I need to do some quick testing to make sure I haven't' broken anything but if that goes well then I'll submit a pull request.  Testing may take me a few days to find time to load KiCad on a Windows machine and to make sure the Windows git isn't messing with the repos (when installing git on Windows, it asks you how you want to handle line endings and the default is to commit Unix style but checkout locally DOS style).  But if anyone else wants to have a look and make any comments or corrections before I submit a pull request I'd be very excited to receive them.

SchrodingersGat, 2017-02-09:
@bradfa thanks for pointing this out. 

I have submitted a PR to address this here: https://github.com/KiCad/kicad-library/pull/988

This is my first time using a .gitattributes file and you seem to have knowledge on them - can you check over the .gitattributes file there?

SchrodingersGat, 2017-02-09:
I have processed the .lib and .dcm files. The .wrl files are taking a long time.

jkriege2, 2017-02-09:
I merged https://github.com/KiCad/kicad-library/pull/988. Maybe we should also add these line-ending requirements to the new KLCs? SHould we do the same for the .pretties?

JAN

SchrodingersGat, 2017-02-09:
Yes I was thinking we need to add this as KLC.

Same for the pretties. It'll be annoying to do this for all the pretty repos (I may have to write a script for that).

(As a side note, I am trying to work on a document to convince the dev team that it's a MUST that all .pretty repos are merged!)

bradfa, 2017-02-09:
@SchrodingersGat if you have a look at https://git-scm.com/docs/gitattributes it gives a good overview of .gitattributes.

@jkriege2 I think it would be good to convert over the .pretties as well, but that's a bit more manual work due to so many repos.  It may not be needed to be added to the KLC as people won't understand because git will just do the right thing if the .gitattributes file is made correctly.

bradfa, 2017-02-09:
@SchrodingersGat there's some nice exact steps at the bottom of https://git-scm.com/docs/gitattributes on how to do the changeover for line endings.  My commit has about 8 million lines changed and 888 files, while yours has a bit less... I worry there's a lot still missing.

SchrodingersGat, 2017-02-09:
I am still working on the other files types, only done .dcm and .lib so far

SchrodingersGat, 2017-02-09:
Here we go https://github.com/KiCad/kicad-library/pull/990

bradfa, 2017-02-09:
@SchrodingersGat you may want to just do

```
*       text=auto
```

And then just call out the binary file name extensions which git gets confused by (like .png and .wings, as they seem to have ASCII data in them near the beginning).  I believe this would do what you want, I just haven't had time to check this on Windows, yet.

SchrodingersGat, 2017-02-09:
hmm good point. I started calling them out individually then the number of file types kept increasing...

MrMartinZockt, 2017-02-12:
> I have processed the .lib and .dcm files. The .wrl files are taking a long time.

If I'm not mistaken, then "library/motor_drivers.dcm" and "library/motor_drivers.lib" are still with CR+LF.

SchrodingersGat, 2017-02-12:
Looks like subsequent PRs have changed line endings back to CR+LF.

I have changed the .gitattributes to enforce LF in https://github.com/KiCad/kicad-library/pull/998

SchrodingersGat, 2017-03-03:
@michalkona was this addressed to your satisfaction? Please close if so :)

closed 2017-03-17 by michalkona