KiCad/kicad-library #819: Trouble meeting Rule 3.6 of KLC

evanshultz, 2016-12-05:
Hi all,

I've been using KiCad for only a little while and found some components that I want which weren't included. I created the parts and ran KLC on them, but I'm not able to solve violations of rule 3.6 on several parts. Surely it's obvious, but I just can't see it and I don't find help here or just Googling. Here are my library files:

[evan.dcm.txt](https://github.com/KiCad/kicad-library/files/630903/evan.dcm.txt)
[evan.lib.txt](https://github.com/KiCad/kicad-library/files/630904/evan.lib.txt)

Using VO615A-2 as an example, the ref des, part type, and pin text is all 50mils. Where am I violating this rule? How to fix it?

Assuming I can get this fixed, I'll create a PR to add these parts to the library. If you have other suggestions or see other problems, please share so that I can learn how to create KiCad parts properly. Thank you!

SchrodingersGat, 2016-12-05:
Hi @evanshultz from the files you have posted, it shows that you also need to change the text size of the extra part fields (which are set to be *invisible* by default)

``'
#
# VO615A-2
#
DEF VO615A-2 U 0 1 Y Y 1 F N
F0 "U" 0 300 50 H V C CNN
F1 "VO615A-2" 0 200 50 H V C CNN
F2 "" 0 0 50 H V L CNN
F3 "" 0 0 60 H I C CNN
'''

See here that the text size of the "F3" field is 60, not 50.

evanshultz, 2016-12-05:
Got it. All valid KLC violations are fixed. Thank you!

closed 2016-12-05 by evanshultz