KiCad/kicad-library #824: RFC: Footprint Libraries

SchrodingersGat, 2016-12-06:
Maybe this suggestion has been raised before, but here goes:

Should we consider merging all the footprint repositories into a single repo, maybe **KiCad/footprints** ?

Managing PRs across dozens of repos (especially keeping any local copies up to date) can get pretty tedious. Having a single repo would also make it much easier for users to submit new PRs.

If we did this, the repository would maintain each current repo as a child directory. Would this allow the github library table to still work? (Obviously with some adjustments required, e.g.

```
(lib (name Capacitors_SMD)(type Github)(uri ${KIGITHUB}/Capacitors_SMD.pretty)(options "")(descr "Capacitors, surface mount"))
```

becomes

```
(lib (name Capacitors_SMD)(type Github)(uri ${KIGITHUB}/footprints/Capacitors_SMD.pretty)(options "")(descr "Capacitors, surface mount"))
```

I think this will vastly improve the way that we handle the github library, with a single repo for symbols, and one for footprints.

( I also think we should have a repo for 3D models but that's a story for a different day )

SchrodingersGat, 2016-12-06:
@pointhi @CarlPoirier @ricardocrudo  @jkriege2 @diggit I'd like your input on this in particular :)

jkriege2, 2016-12-06:
sounds good to me ... currently the footprint maintenance is really a pain.
JAN

SchrodingersGat, 2016-12-07:
Could this be achieved by leaving the current repositories in place and using the submodule architecture? https://github.com/blog/2104-working-with-submodules

keruseykaryu, 2016-12-07:
@SchrodingersGat : "Should we consider merging all the footprint repositories into a single repo, maybe KiCad/footprints ?"

This will not work, because the KiCad's Github plug-in code isn't capable to read repositories from subfolders. At least in time when I tried to do it with my test library.

SchrodingersGat, 2016-12-07:
Hmmm do you see any way forward with this @keruseykaryu ?

keruseykaryu, 2016-12-07:
Don't know. IMO this require completely new Github plug-in code.

bradfa, 2017-02-09:
If you want an easier way to maintain multiple git repos all together without resorting to using submodules or the like, you can take a look at the myrepos tool https://myrepos.branchable.com/

This would allow for the workflow of the librarians to be easier to manage but not impact the workflow of people getting pretty repos from github from within kicad itself.

bobc, 2017-02-27:
(Android repo is another tool https://source.android.com/source/using-repo.html)

However, there is an assumption that the development repo and deployment location must be the same, but that doesn't need to be the case. Normally, binary software updates do not come direct from a source repo.

Assuming
1) because the current github plugin relies on a github trick to get the footprint library and we can't easily change it,
2) we want to have all the good stuff in a small number of places to better maintain it

We can create a master repository (or perhaps one for each type of object), and then use a set of slave repos for deployment. That way we hit both objectives. Creating the deployment repos from the master repo(s) can (and should) be scripted.

There is an additional advantage that the deployment can be done on a more controlled basis, e.g. aligned with KiCad releases.

closed 2017-03-03 by SchrodingersGat