KiCad/kicad-library #825: MOSFET drivers

SchrodingersGat, 2016-12-07:
Currently we have a number of FET driver ICs floating around in various libraries, e.g.

transistors.lib
power_management.lib

and also in manufacturer specific libs.

Does anyone see any advantage in adding a new library (e.g. *mosfet_drivers.lib* for storing these?)

If we can get a quorum on this I will start a PR and try to consolidate the existing symbols.

keruseykaryu, 2016-12-07:
If I could put one's two pennyworth in, you **should** consider **once and for all** how to sort libraries: by function (`passives.lib`, `transistors.lib`, `microcontrollers.lib`, etc.), by manufacturer (`atmel.lib`, `lineartech.lib` `texasinstruments.lib`, etc.) or other key. At present day you have not much libraries to make such a decision and reorganize those which are already present.

IMHO both options generate big and messy libraries which will be a huge problem when Eeschema will support remote libraries like Pcbnew. So it's necessary to find the key which will become the golden mean.

El3ctr0nic, 2016-12-25:
We need fast lib. reorganize..I think. One schema must be approved.

We have fiev manufacturer libs need to rename. (To  keep  "up to date" names)
IR is part of Infineon, Atmel is part of Microchip....
I think that one massive reorganization is needed. We should use function or manufacturer lib name.
Any mix give Us a mess.

Device.lib should be limited to general purpose symbols so
Aruduino, Nucleo, RaspPI headers should have separate library. (lot of my prototype projects have nucleo64/144 as plug in module for easy testing input/output circuits without early drawing uP coo-system drawing).
Template isn't too good for this...it's only my subjective opinion.

Regards,
Marcin

keruseykaryu, 2016-12-25:
@El3ctr0nic: I think that one massive reorganization is needed. We should use function or manufacturer lib name. Any mix give Us a mess.

Maybe it's worth to consider an option to use folders in `share/library`. For example, main folder called according to manufacturer: `Microchip` and several functional libs inside: `Microcontrollers.lib`, `Memory.lib`; `Micron` and `Memory.lib`; etc.

closed 2017-02-27 by SchrodingersGat