KiCad/kicad-library #835: Invalid Atmega part codes

Drakonite, 2016-12-12:
Several of the Atmel Atmega chips seem to have invalid part codes.

An example is KiCad's listing of an ATMEGA168A-P, which as far as I can tell is not listed on Atmel's website and does not exist. It appears it is supposed to be the ATMEGA168A-PU.

This P vs PU part code issue appears to cover most, if not all, of the Atmega chips and their aliases.

I'd be happy to submit a PR that updates the parts to match the valid Atmel part codes if, or if I'm mistaken and the current part codes do/did exist, I can submit a PR that adds the current part codes instead.

Thoughts?

keruseykaryu, 2016-12-12:
A couple of years ago when I start updating `atmel.lib` I assume that single letter suffixes: -P (PDIP), -S (SOIC), -A (TQPF), -M (MLF) will be enough to know "what is what" in this library. The existing parts, before my contribution, also had the same suffixes.

Today, I think KiCad should have a more accurate libraries to attract with them (among the other things) more users. 
It's a lot work, but it return with interest.

Drakonite, 2016-12-13:
> Even if it did generate a large number of aliases.

As long as the -P -S -A -M suffixes were never a part on their own, my thought would be to replace them with the correct -PU -AU etc suffixes, which would not create any additional aliases.

keruseykaryu, 2016-12-13:
IMO If you really like to do this job, do it right and use full Atmel part/ordering codes. For example: the main symbol of ATmega8 TQFP32 will be `ATMEGA8-16AU` and should has few aliases `ATMEGA8L-8AU ATMEGA8L-8AN ATMEGA8-16AN` which are this same MCU but other voltage/speed grades.

Drakonite, 2016-12-13:
I'm going to start by taking a stab at updating the current entries that are broken in ways I'm aware of to match proper Atmel product codes, which mostly seems to be fixing the suffix. It's going fairly quickly so I hope to have a PR later tonight.

I'm running in to a few with invalid product codes, such as ATMEGA8-MI, which seems it's supposed to be ATMEGA8-MU. I'm using my best judgement on them but you'll probably want to double check.

Drakonite, 2016-12-13:
I rescind my "later tonight" comment. I didn't realize how bad/inconsistent the product code naming scheme for the older chips is. I'm having to check datasheets for most things, and I am understanding why this has gone so long without being fixed.  :P

jkriege2, 2016-12-25:
Done!

closed 2016-12-25 by jkriege2