KiCad/kicad-library #848: KLC text anchor proposal

evanshultz, 2016-12-16:
KLC does not specify how and where text should be located or justified. From seeing many library parts and the changes being made (like the huge update from @jkriege2), it seems the following is the correct convention:

`Vertically-oriented symbols have text placed to the right of the symbol which is left-justified. Horizontally-oriented symbols have text above and below the symbol which is center-justified.`

This would also be relatively simple verify in checklib.py as the Y location and justification are easy to parse.

evanshultz, 2017-04-10:
@SchrodingersGat @jkriege2 @poeschlr Submitting for consideration in KLC 2.1...

poeschlr, 2017-04-10:
I think text fields are auto placed by eeschema. (In my neither position nor justification of the text fields in the symbol editor has been taken into account by eeschema)

My test in kicad nightly from october last year:
[image]

[image]

poeschlr, 2017-04-10:
My second test:
[image]

Gasman2014, 2017-04-10:
This is particularly irksome as the default power connections are to the top of the component and this is exactly where the RefDes and name are auto moved to - wherever they happen to be placed in the original library.

[image]

[image]

SchrodingersGat, 2017-04-10:
This is probably worth raising as a [bug-report](https://bugs.launchpad.net/kicad) - the logic that governs the auto-placement of text fields needs serious improvement.

evanshultz, 2017-04-13:
I submitted a very brief bug pointing back to this issue at https://bugs.launchpad.net/kicad/+bug/1682496.

SchrodingersGat, 2017-11-21:
@evanshultz I am closing this due to long time without comment. Please reopen if you feel it is not answered.

closed 2017-11-21 by SchrodingersGat