KiCad/kicad-library #849: Connecting multiple pins with a single pin

evanshultz, 2016-12-16:
Is there a way to have a single logical pin on the schematic symbol connect with multiple physical pins on a footprint?

Here's the example: [Power Integrations' TOPSwitch-JX family](https://ac-dc.power.com/sites/default/files/product-docs/topswitch-jx_family_datasheet.pdf). On page 3 you can see there are three footprints, all with six logical pins but two packages have 6 S pins on the footprint.

Can the schematic symbol have a single S pin for all three packages but that single symbol pin connects to six pins on the footprint if a part is selected that uses a V or K package?

jkriege2, 2016-12-16:
The two ways I know is haveing a footprint with n pads that all have the same number, or you overlay all the pins (with different numbers) in the symbol, then connecting a wire automatically connects to all of them.

Best,
JAN

evanshultz, 2016-12-16:
Thanks for the answer, Jan.

When overlaying pins on the schematic symbol, the "extra" pins are hidden, right? I tried this and it works! :) (I do note that hidden pins are shown in the Choose Compoment dialog which is a bit ugly.)

closed 2017-03-03 by SchrodingersGat