KiCad/kicad-library #869: KLC pin styles and pin lengths

evanshultz, 2016-12-30:
KLC requires pins to be 0.1" in length when pin numbers are (I assume) only two digits (3.2). This works fine for a "Line" graphic style, but not well for an "Inverted" style. There's not enough room to fit an digits comfortably with the circle from the pin style.

Is there a known/favored way to deal with this situation?

When creating a symbol for H11L1, I ran into KLC violations for pin length but I realized that this was very application-specific. The KLC rules are quite reasonable in all cases I can think of, but for some specific parts a unique implementation can be used to meet the pin style and pin length needs. For this symbol, I kept 0.1" pin lengths but added my own 0.05" diameter circle, in effect faking a 0.15" "Inverted"-style pin. Here is what it looks like:
[image]

I think this looks reasonable and fits in nicely with the existing library of parts. However, I'd really like to get input and consider this point with others before submitting this symbol in a PR. Thoughts? Suggestions?

diggit, 2016-12-30:
Take a look into KLC in wiki...
> For black-box symbols, pins have a length of 100mils. **Large pin numbers can be accomodated by incrementing the width in steps of 50mils.**

Use 150mil, 200mil,... As you need. Just place pin connection end on 100mil grid.

diggit, 2016-12-30:
Hmm, i believe, that by incrementing width is meant incrementing pin length. Do you agree @jkriege2, @SchrodingersGat and others?

CarlPoirier, 2016-12-30:
Hi, yes thank you for pointing that out. I will make the correction.

closed 2016-12-30 by CarlPoirier

evanshultz, 2016-12-31:
My question isn't answered, so let me rephrase: Obviously I've read KLC since I mentioned the same clause you quoted, @diggit. Here are the two options I see and their drawbacks:
1. Lengthen the pin as desired even when I don't have a high pin count and keep the type as Inverted. This violates the exact phrase in the KLC, but it's a minor infraction and KLC even states in many places (but not 3.2) that violations can be acceptable but need to be explained when submitting a PR.
2. Make the pin type Line and then draw a fake circle for an inverted pin. This loses some pin information because the type of pin isn't embedded in the part.

I think #1 is the best option, but I gave #2 as an example above since it was rather different than the obvious answer to "lengthen the pin" and didn't violate KLC. Is #1 the correct way to go here? @jkriege2 @SchrodingersGat @CarlPoirier

evanshultz, 2016-12-31:
It appears I cannot re-open this issue, which frustrates me. I, and likely others, would appreciate having the chance to confirm a response before their issues is closed. You cannot be sure the issue is really resolved.

CarlPoirier, 2016-12-31:
Hi @evanshultz,

Indeed #1 is what you need to do. Just think of it as having high enough pin count for the inverted pins to justify lengthening them.

Carl

diggit, 2016-12-31:
@evanshultz 1) is the way to go.

> This violates the exact phrase in the KLC

you mean "Large pin numbers can be accomodated by..."?

It is probably not the best formulation of: "If number does not fit in space that left (by having large number or pin style), increase pin length by _k_\*50mils". (where _k_ is non-negative integer)

evanshultz, 2017-01-01:
All, yes, we're all on the same page now and I'll design in this way in the future. I agree that a small update to KLC could help clarify this case since a high pin number isn't the only case where longer pins have clear value.