KiCad/kicad-library #880: KLC - Symbol Background Fill

SchrodingersGat, 2017-01-03:
Did there (previously) exist a KLC rule stating that symbols should be background-filled with a line-width of 0.01"?

I have been instructing this for a while, after having myself been told to do this for all symbols.

However, now I cannot see this in the KLC - has it been removed or was it an "unspoken" rule?

@CarlPoirier @diggit ?

jkriege2, 2017-01-03:
Hi!

same for me ;-) ... We should definitely have that rule ... I think the check-scripüts enforce it as EC ...

Best,
JAN

SchrodingersGat, 2017-01-03:
Ah, yes the extra checks of course. I feel that these should be actually documented in the KLC

Schlumpf, 2017-01-04:
I agree ;)

SchrodingersGat, 2017-01-04:
Ok so here are the current EC (Extra Check) rules:

## EC01 - Pin types should match function

## EC02 - Pin numbers should contain a numerical character

## EC03 - Position of text fields

## EC04 - Symbol  outline should be 0.01" and filled with background

## EC05 - Duplicate / missing pin numbers

Thoughts:

1. EC01 is worthy of being a rule in its own right
  - This is fairly important and worth more than an "extra check"
  - I think this one certainly should be a proper rule
1. EC02 and EC05 can be merged into a single rule
  - Very similar checks
  - Bullet points under the same rule
  - Other pin-numbering-related checks can be added as further sub points
1. EC04 can be a rule, with other graphical (aesthetic) requirements added
1. EC03 may be merged with 3.8, *or* made as its own rule?

Overall I think that we should remove "extra checks" - either we wish for a rule to be followed or not. I am often asking committers to re-run KLC scripts with the `--enable-extra` flag on.

evanshultz, 2017-01-05:
Agree on all points except that I don't see the correlation between EC03 and KLC 3.8. I would have appreciate knowing about the EC on my first PR as things would have gone smoother/quicker and I could have learned what needed to be done up front.

SchrodingersGat, 2017-01-05:
3.8 was a bit tenuous - probably best as its own rule

SchrodingersGat, 2017-01-16:
Closing this as it has been addressed here - https://github.com/KiCad/kicad-library/issues/879

closed 2017-01-16 by SchrodingersGat