KiCad/kicad-library #882: KLC-rule 4.4

jkriege2, 2017-01-04:
Hi all.

as we are currently reworking some rules ... here's one more:

> 4.4 Any modification to the original symbol, indicated by appending the reason. Ex: different pin ordering: "Transistor_PNP_Pinswap1".

I think that is not really done anywhewre, but we have e.g. transistors with naming `Q_NPN_EBC` and `Q_NPN_BCE`, ... to indicate the pin ordering, So I would suggest something like this:

> 4.4.NEW If several devices exist that only differ by pin ordering (e.g. NPN transistors with different pinouts), this should be made clear in the symbol name, e.g. `Q_NPN_EBC`, `Q_NPN_BCE`, ...

What do you think?
JAN

SchrodingersGat, 2017-01-04:
Good idea - see https://github.com/KiCad/kicad-library/issues/879

SchrodingersGat, 2017-01-12:
Closing this as it has been added to #879

closed 2017-01-12 by SchrodingersGat