KiCad/kicad-library #884: new KLC rule 3.10 for footprint filters

jkriege2, 2017-01-04:
Hi again!

In my experience the footprint filters are a real mess and nowhere defined properly. SO I would propose a new rule:

> 3.10 Whenever possible footprint filters should be set that allow to match all packages that can be used for a symbol (also e.g. different types of packages with the same pinout, such as SOIC-8, TSSOP-8, DIP-8 ...). The footprint filters should follow these rules:
>  - FPFilters use wildcards (`*` matches an number of charactzers and `?` to match 0 or 1 character)
>  - FPFilters should end with a `*`-wildcard to allow matching different modifications of a package (e.g. `_HandSoldering` or `_LargePads`)
>  - FPFilters do *not contain the pin count*. It is matched by the pin count filter of KiCAD
>  - FPFilters should match hyphens ... by `?`, e.g. `TO?220?5*` to match a `TO-220-5`
>  - FPFilters should match as much dimension information as possible to be as specific as possible, e.g. `DIP*W7.62mm*` to match `DIP-22_W7.62mm` and not a DIP-22 with different width, or `SOIC*3.9x8.7mm*Pitch1.27mm*` to match `Housings_SOIC:SOIC-14_3.9x8.7mm_Pitch1.27mm`

What do you think?

JAN

SchrodingersGat, 2017-01-04:
I have implemented `3.10` in https://github.com/KiCad/kicad-library/issues/879

I think that `3.11` and `3.12` should be consolidated into a single rule "Pin Numbering" or something like that - and specified as bullet points. There are a number of other rules that could be included regarding pin numbering

SchrodingersGat, 2017-01-04:
@diggit  both good points - see changes to https://github.com/SchrodingersGat/kicad-library/wiki/KLC

SchrodingersGat, 2017-02-23:
Closing this as it is dealth with in #879

closed 2017-02-23 by SchrodingersGat