KiCad/kicad-library #885: KLC and text width

jkriege2, 2017-01-04:
Hi!

one thing I noted when playing around with the KLC scripts:
Currently we require a linewidth of 0.12mm for Courtyard and 0.1mm for Fabrication layer, but text on both should have a thickness of 0.15mm ... So:
1. Should we match the text linewidth to the linewidth on the respective layer, or leave it as it is (which looks good to me from an ästhetic standpoint!)
2. should we mention this in the rules?

JAN

tmanaud, 2017-01-06:
About courtyard, what is the point to have a 0.1mm as it's fairly big for 0201.
As it's not intend to be printed or use anywhere else Pcbnew why not reducing it?
I've been using 0.01 for a while, and it's perfect.

evanshultz, 2017-01-16:
@jkriege2 Good catch.
1. I like leaving it as is for the same reason.
2. Yeah... otherwise it's just tribal knowledge and new users will struggle. Perhaps it can also be captured in the updated check_kicad_mod.py to also help conformance to KLC?

@tmanaud You must mean the courtyard grid. Courtyards are fairly nebulous, in my experience, because they often mean different things for the board designer, librarian, manufacturing assembly engineer, solder/process engineer, etc. There are probably as many corner cases as there are part packages, but it's a line in the sand to show an outer boundary of the entire land pattern. But again, what this means or why it has a specific set of rules can vary. While you're free to do whatever you like on your own symbols, for inclusion into the official KiCad library we need to conform with the KLC rules where possible (and also continuously review the rules to make sure they're modern and reasonable).

SchrodingersGat, 2017-03-03:
Closing this as it will be dealt with in KLC 2.0 (upcoming)

closed 2017-03-03 by SchrodingersGat