KiCad/kicad-library #890: No coupled inductor in new device.lib

evanshultz, 2017-01-07:
In PR #742 (thanks @jkriege2!) I don't see a coupled inductor symbol. This would be useful for a [Cuk converter](https://en.wikipedia.org/wiki/%C4%86uk_converter), for example, where the inductors aren't logically a common mode choke. While the schematic symbol could work, for these applications I believe the schematic looks much better with two separate symbols.

Does anyone else agree? Is there already a coupled inductor which I could use? If not, I'll be happy to create one and submit a PR.

evanshultz, 2017-01-11:
Any input? Both into the concept or using a single unit vs two units (my preference) for this symbol?

Coupled inductors may have a core or may not, so the following would seem to be needed to match all the varieties in the existing device.lib:
- `L_Coupled`
- `L_Core_Ferrite_Coupled`
- `L_Core_Ferrite_Coupled_Small`
- `L_Core_Iron_Coupled`
- `L_Core_Iron_Coupled_Small`
- `L_Coupled_Small`

jkriege2, 2017-01-11:
Did you also have a look ate the `Transformer...`-synbols?
I would prefer using a single unit, as otherwise it is not always clear, which pair belongs together ... especially since the two separate subunits are not independent and exchangeable ...!

Best,
JAN

evanshultz, 2017-01-13:
I did, any I really like separate inductor symbols. Recall that, traditionally and typically, a transformer doesn't store energy while an inductor does (this distinction isn't always true, for example with a flyback transformer). But having a coupled inductor part, with separate symbols for cleanest schematic design (see the above schematic example), appeals to me.

Using two units forces 'A' and 'B' suffixes (or other suffix style, if chosen by the user) on the ref des, plus having pin numbers shown, plus having dots shown on both symbols, seems quite clear to me. I can't see how anyone skilled enough to use KiCad wouldn't know this or couldn't find the answer.

For example, see this symbol:
```
# INDUCTOR_COUPLED
#
DEF INDUCTOR_COUPLED L 0 0 N Y 2 L N
F0 "L" 0 125 50 H V C CNN
F1 "INDUCTOR_COUPLED" 0 -50 50 H V C CNN
F2 "" 0 0 50 H V C CNN
F3 "" 0 0 50 H V C CNN
$FPLIST
 Choke_*
 *Coil*
$ENDFPLIST
DRAW
A -150 0 50 1 1799 0 1 0 N -100 0 -200 0
A -50 0 50 1 1799 0 1 0 N 0 0 -100 0
A 50 0 50 1 1799 0 1 0 N 100 0 0 0
A 150 0 50 1 1799 0 1 0 N 200 0 100 0
C -150 0 10 0 1 0 F
P 2 0 1 0 200 75 -200 75 N
X A1 1 -300 0 100 R 50 50 1 1 I
X A2 2 300 0 100 L 50 50 1 1 I
X B1 3 -300 0 100 R 50 50 2 1 I
X B2 4 300 0 100 L 50 50 2 1 I
ENDDRAW
ENDDEF
```

SchrodingersGat, 2017-01-13:
Perhaps having both a single and a split would be ok?

evanshultz, 2017-01-15:
I'm fine having both. My question, before I build the symbols, is would they be merged? No sense in building symbols that aren't going to be merged if I want to contribute to the default libraries.

jkriege2, 2017-01-15:
Why not? If we agree on having both options ... I don't see why not ;-)

PS: Thinking about these ... please also have a look at `EMI_FILTER_LL` in `device.lib`:
[image]
Maybe you can also work with that, or if you change the style a bit make sure the style of this (and the other inductors in the lib) are matching up ...

cu
JAN

SchrodingersGat, 2017-01-16:
@evanshultz your work would not be wasted ;)

jkriege2, 2017-01-19:
For the naming I would try and have all of the as `L_Coupled...` ... and one note: With the new inductors-library you could use this FPFilter:

> $FPLIST
>  L_*
>  L_CommonMode*
> $ENDFPLIST

JAN

SchrodingersGat, 2017-02-17:
@evanshultz have these symbols been integrated into the libraries now?

evanshultz, 2017-02-18:
Not yet. I will be working on them in the next couple days. Thanks for checking in.

evanshultz, 2017-02-24:
Here are a smattering of items to consider:
- On the left is the original split symbol I made
- Top middle is the "L" symbol in the current library
- Below it is a coupled version (without the line for the core, yet):
- On the right is a split version of "EMI_Filter_LL"
- And below that is a renamed version
[image]

None of these are finished and some have overlapping elements, etc. Just talking points to pick a direction.

Thoughts? Personally, I like the bigger part on the left but I could go for the EMI_Fiter_LL version too since it more closely matches what's in device.lib now.

SchrodingersGat, 2017-11-21:
@evanshultz has this been answered to your liking? Please close if so.

evanshultz, 2017-11-21:
No, I still plan to do this once I finish the pot script and get those footprints updated.

poeschlr, 2018-01-20:
@evanshultz and @jkriege2 if this is still relevant please reopen over at the new repo

closed 2018-01-20 by poeschlr