KiCad/kicad-library #891: images to illustrate KLCs

jkriege2, 2017-01-07:
## Box symbols
[image]
[image]
[image]

## NC-pins, pin-length
[image]

## pin-grouping + left/right
[image]

## pin types
[image]

## symbol properties
[image]
[image]

## symbol anchor
[image]

## THT package
[image]

## SMD package
[image]

## footprint properties
[image]

cvgaviao, 2017-01-07:
There is a typo in the description of GND pin of the first image. it is bottom instead top, right?

jkriege2, 2017-01-07:
@cvgaviao thanks: fixed it!

diggit, 2017-01-08:
Image 2 shows counter clockwise port ordering, but ports on right side are not. It is (from top) CDE, should be EDC.

diggit, 2017-01-08:
I am not telling that CCW rule is better, the way it is makes more sense IMO.

nickoe, 2017-01-14:
FWIW, it does no make sense to force a port ordering strictly. I mean, sometimes groups of pins is better grouped, but the numbers might not align up properly. But I guess if you have a stretch of GPIO pins or similar not so different pins, you should try to order dem CCW for example.

jkriege2, 2017-01-15:
that's how I understand the rule ... I wouldn't enforce it only for the sake of the rule ... really has to make sense in the actual case ;-)

JoanTheSpark, 2017-01-29:
If the CCW rule has got such a low priority I would make that clear in the pictures though (grey color for it's bubble & text like the NC pins maybe?).

Anyhow, good work!

jkriege2, 2017-01-29:
I updated the image on the CCW-rule (which was removed anyways):
[image]

pointhi, 2017-01-29:
@jkriege2 

I think the pin length on your image is 150mil, not 50mil (the text size should be 50mil)

What about doing a left align of reference and value for symbols like ic's, and matching the left start point of those references. (Note: Those references are not static, but can increase in size, and such behaviour would prevent overlap of references and pins near it)

Like:

[image]

It's only a proposal, but it would be good to create a "perfect" example symbol :)

Thanks for creating such greate images, which should help newbies quite a bit.

sujitth, 2017-01-29:
Hi,
is there any rules about the Pin name position offset?

[image]

a value of 15 or 20 will be comfortable for symbols which having internal drawings.

Thanks for the great reference image.

Best Regards,

Sujith

SchrodingersGat, 2017-01-29:
Jan,

This is some fine work. Having images will really help users to understand the KLC.

I have started to add some of these to the updated [KLC](https://github.com/SchrodingersGat/kicad-library/wiki/KLC). Refer to 3.10 / 3.12 / 8 / 9 for instances of how I think they should be integrated.

Images are placed in "down-down" boxes. This works on Chrome really well. Can we get some testing on other browsers too? e.g.

[image]

I also have some suggestions as to visual style for the images:

## Multiple Rules
Some of the images cover too many rules at once (is confusing!) Some can (should) be split into separate images, one for each rule.
However, some may work better as addressing multiple rules in the same image. For this, I would suggest that each callout contains the number of the associated rule. e.g.
[image]

jkriege2, 2017-01-29:
Hi! I'll work a bit on the images ... I just checked the way you included them ... hhmmm ... the problem (at least on firefox) for me is that it is absolutely not obvious that you can open thise things up ... wouldn't it be possible to simply have the image-thumbnail floating on the right (e.g. as on Wikipedia)

JAN

jkriege2, 2017-01-29:
# Image for KLC 8.2
[image]

jkriege2, 2017-01-29:
# Image for KLC 9.2+9.3
[image]

jkriege2, 2017-01-29:
# image for KLC 7.3 and 7.4
[image]

# image for KLC 7.5 and 7.6
[image]

jkriege2, 2017-01-29:
# Image for KLC 3.x
[image]

jkriege2, 2017-01-29:
## Another image for KLC 3.x
[image]

jkriege2, 2017-01-29:
## Annotated image for KLC 10 (and 9.1)
[image]

jkriege2, 2017-01-29:
## Annotated image for KLC 10.x and 8.1 (SMD)
[image]

jkriege2, 2017-01-29:
## Image for SMD components
[image]

CarlPoirier, 2017-01-29:
Hi @jkriege2,

These are great. May I ask with what program you did them? It would be great to commit the sources to allow modifications further down the road.

jkriege2, 2017-01-29:
Hi1

I used http://getgreenshot.org/ which is a really nice screenshot tool that allows these annotations. I can provide these files (just how to share them)? I cold upload them to my webpage and post a link, or I simply upload them somehwere to the repos when we#re done1

JAN

SchrodingersGat, 2017-01-30:
Ok I have added these images (inline, now - https://github.com/SchrodingersGat/kicad-library/wiki/KLC). 

Another option is to just link them:

[Example Image](https://cloud.githubusercontent.com/assets/11967523/22405545/cafd0006-e644-11e6-85e3-b600c860cfad.png)

What do people think? The drop-down box only really works in Chrome so that's not really an option.

jkriege2, 2017-01-30:
Hi!

I like it if they are inside the text (if thumbnails are not possible ...)

I made a PR to have the images in the repo, so you can link against them without having to rely on the githubcontent.com ...

https://github.com/KiCad/kicad-library/pull/961

Please merge if you think that's a good solution. the .greenshot files are also there so one can still edit the text labels ...

Best,
JAN

SchrodingersGat, 2017-01-30:
Turns out that you can just use raw HTML so now there are thumbnails - https://github.com/SchrodingersGat/kicad-library/wiki/KLC

poeschlr, 2017-02-25:
The screenshots using the dip 8 package are missing the pin 1 marker on silk.

Both the smd and tht screenshot miss the reference on the fab layer.

SchrodingersGat, 2017-03-05:
Thanks for these images :)

closed 2017-03-05 by SchrodingersGat