KiCad/kicad-library #892: make 74xx library KLC compatible and extend with more devices

koder68, 2017-01-07:
The 74xx.lib library today is far from KLC compliant. It should be updated to follow the rules. Also there are a lot of devices that are not included. For some I have sen there are issues entered but not for those I needed.

I could volunteer to do at least some of this work but as far as I understand some of the changes needed will break existing schematic diagrams and I kind of seek approval to go ahead anyway.

- Changing pin lenghts.
- Changing origo
- Changing description to match these from wikipedia (https://en.wikipedia.org/wiki/List_of_7400_series_integrated_circuits)
- Moving label and reference to some common location (but which)
- Using keywords from kind of a controlled vocabulary
- Making the 74xx part root for all components and HC, HCT, L, S, LS, ALS and F variants just aliases.

It would be a lot of work and I guess some use the ttl_ieee library which I plan not to touch.

One question would be what to do with the 1-gate devices. There is a separate 74xgxx library. Perhaps those in the 74xx library should be moved there.

koder68, 2017-01-08:
After some checking i'm not overly enthusiastic about this .1 inch standard pin length. For pins with inverted input its seldom enough so need to go to .15 and that is not enough for pins with 2 digit numbers. .2 inch works but why have .1 inch on one side and .2 on the other. And why have .1 inch on one device and .2 on another.
All in all, I would like to make every pin for black box components .2 inch long. Besides, the checker does not seem to check pin length...

jkriege2, 2017-01-11:
Hi!

I think for the 74xx-libs, a lot more work has to be done (maybe a complete rewrite would be needed), because there is also the topic of the hidden power-pins, which many think should be removed, see: https://github.com/KiCad/kicad-library/issues/605

Best,
JAN

koder68, 2017-01-11:
I had seen that issue but obviously not read it in detail. I have now and will think about this for a while and then add to that discussion.

bobc, 2017-06-26:
Just an FYI, I am working on an updated 74xx.lib. I would  like to address several of the issues mentioned here.

Many of the logic devices no longer have a 'standard' part available, e.g. 7400,7402 and are only available in HC, HCT etc. However, I think it will be a lot tidier if the base part is always 74xx and a lot easier to search.

I have scraped the data from the wiki page into a text file with a simple script, so I can automatically add descriptions, datasheets. Suggestions for standardised keywords welcome!

SchrodingersGat, 2017-11-21:
Closing as the new logic work done by bobc has been merged now.

closed 2017-11-21 by SchrodingersGat