KiCad/kicad-library #893: Hide Footprint field for All Components in All KiCAD libraries

kaitrek, 2017-01-09:
It does not make sense why anyone would want to show the footprint field in the schematic.
Now that KiCAD is implementing the new footprint identification system where library_name:footprint_name is VERY long, it will clutter up the schematic even more and force users to unnecessarily uncheck the Show checkbox for every component in their schematic.

Someone should run a batch operation to hide the footprint field for all components in kicad libraries.

cpavlina, 2017-01-09:
Agreed 100%, the footprint names are ridiculously long now and nobody wants to see those all over their schematic.

keruseykaryu, 2017-01-09:
I disagree. For example I want to see clearly footprint in Library Browser and when I select symbol to place.
The goal is not to hide it in the library because it is important information, but globally show/hide it by the user on the schematic. The code already exist - Eeeschema show a dialog and ask what to do with footprint fields after back annotate. The only thing is to connect this to some toolbar button or menu command to call this code at user demand.

jkriege2, 2017-01-09:
I wold also like to have the fields hidden ... as said ... after back-annotate you can switch them on/off all at once! Also: We now usually add the name of the package to the part-description (at least where it makes sense, e.g. for ICs that come in different flavours), so this info is often available in the lib browser:
[image]

Maybe this could be even taken further: The symbol-selector already shows whether this is an Alias, shows the keywords, ... so why not also show the package there?

@keruseykaryu Still having such a button would be a nice addition ... are you planning on implementing that?

Best,
JAN

keruseykaryu, 2017-01-09:
Sorry @jkriege2 my KiCad role has been self switched down from "involved user" to "basic user".

cpavlina, 2017-01-09:
Perhaps the component selector should show all fields even if they are hidden, the same way the library editor does.

SchrodingersGat, 2017-01-09:
I would even suggest displaying the footprint field explicitly:

[image]

evanshultz, 2017-01-13:
@kaitrek To directly answer you question: yes, they should. This is covered by rule 3.9 of the [KiCad Library Convention](https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention). Anywhere you don't see this happening, it's because the library is old and not yet been updated. This is a slow process but should eventually happen.

And I agree with the discussion... hide them on the schematic page by default, but show the info during component selection.

SchrodingersGat, 2017-01-16:
PR here https://github.com/KiCad/kicad-library/pull/917/files

SchrodingersGat, 2017-01-21:
This has been addressed :)

closed 2017-01-21 by SchrodingersGat