KiCad/kicad-library #894: Hide Datasheet field for All Components in All KiCAD libraries

kaitrek, 2017-01-09:
Same as issue #893 but with Datasheet field. No one needs to see the datasheet http link on the schematic for every component...

Suggest the Datasheet field be hidden by default.

evanshultz, 2017-01-13:
@kaitrek Indeed, they should be hidden. I've not yet come across any components where the datasheet is set to visible, but perhaps you're using a part I haven't used before. (Though I'm using parts with the datasheet definitely present in the library's DCM file.) Can you please share which part or parts have a visible datasheet link so we can see what you're seeing?

SchrodingersGat, 2017-01-16:
PR here https://github.com/KiCad/kicad-library/pull/917/files

SchrodingersGat, 2017-01-21:
This has been sorted :)

closed 2017-01-21 by SchrodingersGat