KiCad/kicad-library #900: Fuse symbol in device.lib

evanshultz, 2017-01-11:
The current (newest) device.lib shows only a small box with a line through it for a fuse. IEEE Std 315-1975 section 9.1.1 shows this as one usable fuse symbol, but also a small box with lines at each of the small sends and a pair of arcs in series.

@jkriege2 why did you only select this one symbol? This symbol is marked in the IEEE doc as "IEC" so perhaps it was the most commonly accepted and so deemed the best single symbol for KiCad?

I personally prefer the double arc symbol, but I think this is regional. Would anyone object to me adding them back into the library, perhaps as `Fuse_Alt` and `Fuse_Alt_Small`?

jkriege2, 2017-01-11:
This is the standard IEC-symbol ... and we try to follow IEC ... therefore we also don't have the wavy resistor symbol, but the box ...

evanshultz, 2017-01-11:
That's what I guessed; thanks for confirming. So the wavy symbol wouldn't be accepted into the library?

SchrodingersGat, 2017-01-16:
We will stick with the standard (IEC) symbol rather than having both.

evanshultz, 2017-01-16:
Well, there you have it.

closed 2017-01-16 by evanshultz