KiCad/kicad-library #901: How is 'Status of the Libraries' maintained?

evanshultz, 2017-01-11:
https://github.com/KiCad/kicad-library/wiki/Status-of-the-libraries seems to be out of date? @jkriege2 just recently merged a big PR from me that made powerint.lib KLC-compliant, but the wiki already had it marked as such. I suspect other library information is also outdated.

@keruseykaryu, you seem to be the last one to update the page but that was quite some time ago. @SchrodingersGat? @CarlPoirier?

keruseykaryu, 2017-01-11:
@evanshultz I'm not a library maintainer anymore. So anybody else in current repo maintainers team should take care of this (or remove).

SchrodingersGat, 2017-01-11:
@evanshultz thanks for pointing this out. I think this table should be updated, and also include a description of the contents of each library. This will also help users when deciding where to put new symbols.

Hopefully I'll have a chance to look at this some time this week.

evanshultz, 2017-01-12:
That sounds very good. If you need library maintainers, I'm happy to be considered. With Jan's help we got powerint.lib updated and there are a few other libraries that I'm passionate about.

SchrodingersGat, 2017-01-12:
@evanshultz noted 👍

SchrodingersGat, 2017-01-12:
A start has been made on updating this:

https://github.com/KiCad/kicad-library/wiki/Status-of-the-libraries

SchrodingersGat, 2017-01-12:
I would also like to see a similar table documenting the status of the official .pretty libs

CarlPoirier, 2017-01-13:
These tables could go in the Wiki.

SchrodingersGat, 2017-01-13:
@CarlPoirier do you mean as another top-level wiki page? That is my thinking too. A page titled "Footprint Libraries" with an up-to-date table with links, etc.

Plus links to scripts for downloading / updating all the libs. The current scripts (are they still buried in the source?) are pretty out of date.

CarlPoirier, 2017-01-13:
@SchrodingersGat Yes, sounds good.

@evanshultz Please send me an email so we can discuss about you becoming a maintainer. You will find my email in kicad-library commit history with "git log | grep carl".

evanshultz, 2017-01-13:
@CarlPoirier This looks much better! Would it be prudent to add a column for the maintainer of each library so folks know some person to contact/mention if they have a question/problem?

CarlPoirier, 2017-01-21:
@evanshultz If you can find a maintainer for each library, I think it could be great.

yo3iti, 2017-02-20:
Hello everybody, Curious About the status of this. Is the an "official" maintainer ? :) BR

SchrodingersGat, 2017-02-20:
Hi @yo3iti currently I am the one charged with keeping the library team on task ;)

yo3iti, 2017-02-20:
Hi, @SchrodingersGat I could help. I suggest a proof-of-concept. Let me share with you what I did for Analog_devices library. Maybe this will be accepted for the management of the rest of libraries.

closed 2017-09-28 by SchrodingersGat