KiCad/kicad-library #910: SOT23 single diodes do not work with the SOT-23 KiCad footprint

evanshultz, 2017-01-15:
I think I found a notable problem where KiCad is 'wrong'. See these parts:
- BZX84 series: http://www.nxp.com/documents/data_sheet/BZX84_SER.pdf
- BAS19/20/21: http://www.diodes.com/_files/datasheets/ds12004.pdf
- IMBD4148: http://www.vishay.com/docs/85731/imbd4148.pdf

(Below I'll stick with the SOT-23 pinout in KiCad, where pins 1 and 2 are along the same side and pin 3 all by its lonesome.)

You'll notice that all of them would physically fit on the `TO_SOT_Packages_SMD:SOT-23` footprint and have electrical connections to pins 1 and 3. However, the symbol connects to pins 1 and 2. Currently, selecting the SOT-23 package, if the user has this diode in mind, means cathode of the diode is going to the wrong package pin!

You can see that this is a common package for many diodes from many manufacturers.

Since making alternate schematic symbols seems to be more work for the librarians, and makes the user think more before they get to CvPcb, I an alternate SOT-23 footprint named `TO_SOT_Packages_SMD:D_SOT-23`. This footprint would swap the locations of pins 2 and 3 from the existing SOT-23 footprint and have a clear naming scheme with the `D_` prefix. A `_Handsoldering` version should be created as well.

If this is all good, I'll get right on it. :)

jkriege2, 2017-01-15:
Hi!

hhmmm ... somehow both, the current state and the suggested solution are not perfect:
- currently you'll have to derive for every diode anew
- with the proposed change we have a SOT-23 that does not matche the pin-numbers usually used in the datasheets (which is a constraint I understand we want to impose).

Therefore I would suggest a different solution: To match the different variants of the diodes we already have I would therefore suggest to make variants `D_Schottky_ANK` and `D_ANK` ... which would have the pin 1=A, pin2=NC+hidden and pin3=K ... + possibly some rotations of needed ...

What does everybody else think?

JAN

SchrodingersGat, 2017-01-16:
For my 2c I think that a good approach would be to add a SOT-23_ANK to the https://github.com/KiCad/Diodes_SMD.pretty as this is a common, diode-related problem. This way, we can   also conform to the existing diode-related footprint filters, and users won't be presented with weird packages in the more-generic TO-SOT library.

evanshultz, 2017-01-16:
What I don't like about that is that there will need to be a unique diode schematic symbol for this package. From a user's perspective, I'd like to pick a diode and then have free reign to choose any package later. This is a big thing that differentiates KiCad since a symbol and footprint aren't locked together.

To do that, however, we need to tie pins 1 and 2 of the existing diode symbol to pins 1 and 2 on the footprint while also having a dummy pin. Is there a way to do that? I tried naming a footprint pin `N`, `~`, and a few other things but CvPcb always knows I'm trying to match a 2-pin symbol with a 3-pin footprint.

Is there a way to make this (in my opinion) clever and smooth method work?

SchrodingersGat, 2017-01-16:
[image]

It looks like it works if you just leave the pad blank - the footprint then  comes up in CvPcb when you click "filter by pin count"

evanshultz, 2017-01-16:
Oh. Yeah. Indeed it does. Not sure what I was doing wrong...

So would it be acceptable to add `Diodes_SMD:D_SOT-23` as follows:
```
(module D_SOT-23 (layer F.Cu) (tedit 587C4E18)
  (descr "SOT-23, Single Diode")
  (tags SOT-23)
  (attr smd)
  (fp_text reference REF** (at 0 -2.25) (layer F.SilkS)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_text value D_SOT-23 (at 0 2.3) (layer F.Fab)
    (effects (font (size 1 1) (thickness 0.15)))
  )
  (fp_line (start -1.65 -1.6) (end 1.65 -1.6) (layer F.CrtYd) (width 0.05))
  (fp_line (start 1.65 -1.6) (end 1.65 1.6) (layer F.CrtYd) (width 0.05))
  (fp_line (start 1.65 1.6) (end -1.65 1.6) (layer F.CrtYd) (width 0.05))
  (fp_line (start -1.65 1.6) (end -1.65 -1.6) (layer F.CrtYd) (width 0.05))
  (fp_line (start 1.29916 -0.65024) (end 1.2509 -0.65024) (layer F.SilkS) (width 0.15))
  (fp_line (start -1.49982 0.0508) (end -1.49982 -0.65024) (layer F.SilkS) (width 0.15))
  (fp_line (start -1.49982 -0.65024) (end -1.2509 -0.65024) (layer F.SilkS) (width 0.15))
  (fp_line (start 1.29916 -0.65024) (end 1.49982 -0.65024) (layer F.SilkS) (width 0.15))
  (fp_line (start 1.49982 -0.65024) (end 1.49982 0.0508) (layer F.SilkS) (width 0.15))
  (pad 1 smd rect (at -0.95 1.00076) (size 0.8001 0.8001) (layers F.Cu F.Paste F.Mask))
  (pad "" smd rect (at 0.95 1.00076) (size 0.8001 0.8001) (layers F.Cu F.Paste F.Mask))
  (pad 2 smd rect (at 0 -0.99822) (size 0.8001 0.8001) (layers F.Cu F.Paste F.Mask))
  (model TO_SOT_Packages_SMD.3dshapes/SOT-23.wrl
    (at (xyz 0 0 0))
    (scale (xyz 1 1 1))
    (rotate (xyz 0 0 0))
  )
)
```

That seems easiest for users since the existing diode symbols don't need changed (I tried `D`, `D_Schottky`, and `ZENERsmall`), but using the aforementioned symbols their footprint filters may need updated to avoid blocking otherwise valid diode footprints.

SchrodingersGat, 2017-01-16:
Almost, with a few caveats:

1. I think it would need a suffix showing which pins are being used e.g. D_SOT-32_CNA (if pin-1 is indeed Cathode)
2. Copy the 3D shape into Diodes_SMD.3dshapes (in the future, I hope that the 3D models will be packaged with the footprints)

Submit this as a PR (to https://github.com/KiCad/Diodes_SMD.pretty) and we'll go from there :)

evanshultz, 2017-01-16:
Will do. Closing this now...

closed 2017-01-16 by evanshultz