KiCad/kicad-library #933: TVS-specific diodes are bidirectional

evanshultz, 2017-01-18:
@jkriege2 I think this may be mostly for you since you put in so much work to update device.lib.

The diodes in that lib named "TVS" are only a bidirectional TVS symbol type, but it's equally valid to have a unidirectional TVS. The unidirectional one would normally just use a zener diode symbol. See [ST SM6T](http://www.st.com/content/ccc/resource/technical/document/datasheet/6f/2d/a6/9e/56/51/48/15/CD00000725.pdf/files/CD00000725.pdf/jcr:content/translations/en.CD00000725.pdf) for an example.

Is it best to update the part name or just the description in the DCM files for these TVS symbols to clearly indicate that they're only a symbol for bidirectional TVS diodes?

jkriege2, 2017-01-18:
yes, but then one could simply use `D_Zener`? Go ahead and modify the description, I think that'S enough ... as it is obvious from the symbol?

But that's just my 2cents ...  @pointhi @SchrodingersGat @diggit what do you think?

SchrodingersGat, 2017-01-18:
I'm happy for the description to include the word *bidirectional* or similar. Zener symbol should suffice for unidirectional TVS.

evanshultz, 2017-01-19:
PR #939 submitted.

closed 2017-01-19 by evanshultz