KiCad/kicad-library #956: Question about 74LVC1G125

penoud, 2017-01-26:
How to do this component
http://www.nxp.com/documents/data_sheet/74LVC1G125.pdf 
regarding alias convention. Did I have to do 7 component with the footprint in the name, such as 74LVC1G125_SOT353-1, ..., 74LVC1G125_SOT1226 ? 
Or did I do 2 component 
1) 74LVC1G125_SOT353-1
2) 74LVC1G125_SOT886
add alias for the same pin configuration but with a less clear filter for the footprints

jkriege2, 2017-01-26:
I would make two symbols, one for the 5-pin packages and one for the 6-pin packages. Then make FPFilters for each that match all matching packages. e.g.
- `SOT-886*`
- `SOT-891*`
- `SOT-1115*`
- `SOT-1202*`
... and then call the two symbols maybe `74LVC1G125G_5pinPack` and `74LVC1G125G_6pinPack`
... or even make one symbol with exact footprint-name in footprint-field and FPFilter for each package, then using the exact part number from table 1 in the DS.

JAN

penoud, 2017-01-27:
Thank you for you clear explanation. 
Now, go for work!

jkriege2, 2017-01-27:
One more comment ... the logic libs are currently in awfully bad shape ... 

- https://github.com/KiCad/kicad-library/issues/892
- https://github.com/KiCad/kicad-library/issues/605

penoud, 2017-02-02:
Thank you I read carefully #892 and #605. It seems no census is found yet. I would be pleased to help for helping in this issue.

closed 2017-02-02 by penoud