KiCad/kicad-library #984: Case sensitivity - Windows

MrMartinZockt, 2017-02-08:
There are two libraries called NXP. One in uppercase and one in lowercase.
But Windows has a case insensitive filesystem. So it is not possible to clone the library properly. One file is always missing.

SchrodingersGat, 2017-02-08:
Thanks for pointing this out @MrMartinZockt . I have deleted NXP.lib/.dcm (upper case). I will merge the components that were in the upper-case lib into the lower-case one.

SchrodingersGat, 2017-02-08:
Fix is here - https://github.com/KiCad/kicad-library/pull/985

jkriege2, 2017-02-08:
Done!
Thanks for pointing this out!

MrMartinZockt, 2017-02-08:
I found another one

[modules/packages3d/unused_3d.3dshapes/device.douille_4mm(black).wrl](https://github.com/MrMartinZockt/kicad-library/blob/master/modules/packages3d/unused_3d.3dshapes/device.douille_4mm(black).wrl)
[modules/packages3d/unused_3d.3dshapes/device.douille_4mm(black).wings](https://github.com/MrMartinZockt/kicad-library/blob/master/modules/packages3d/unused_3d.3dshapes/device.douille_4mm(black).wings)

and 

[modules/packages3d/unused_3d.3dshapes/Device.douille_4mm(black).wrl](https://github.com/MrMartinZockt/kicad-library/blob/master/modules/packages3d/unused_3d.3dshapes/Device.douille_4mm(black).wrl)
[modules/packages3d/unused_3d.3dshapes/Device.douille_4mm(black).wings](https://github.com/MrMartinZockt/kicad-library/blob/master/modules/packages3d/unused_3d.3dshapes/Device.douille_4mm(black).wings)

SchrodingersGat, 2017-02-08:
Good find! These have been fixed also.

MrMartinZockt, 2017-02-08:
Thanks!
Everything is working fine now. :+1:

closed 2017-02-08 by MrMartinZockt