KiCad/kicad-library #986: Alternate symbol for dual diodes

evanshultz, 2017-02-08:
While the "opposed diode" symbol is nice in some cases, for others I find it unwieldy, For example, when using a common cathode rectifier on the output of a flyback supply. Having the diode symbols shown in parallel, next to each other, is more attractive to me.

Here are some example parts that, to me, better fit the symbol described above:
- [STPS40L15CT](http://www.st.com/content/ccc/resource/technical/document/datasheet/a3/ec/15/c3/8c/15/4b/98/CD00001105.pdf/files/CD00001105.pdf/jcr:content/translations/en.CD00001105.pdf)
- [MBR20200CT](http://www.onsemi.com/pub_link/Collateral/MBR20200CT-D.PDF)

Here is the symbol:
[image]

And the symbol lib instance:
> `# D_Schottky_x2_KCom_AKA_Alt
> #
> DEF D_Schottky_x2_KCom_AKA_Alt D 0 40 Y N 1 F N
> F0 "D" 0 250 50 H V C CNN
F1 "D_Schottky_x2_KCom_AKA_Alt" 0 -250 50 H V C CNN
F2 "" 100 -50 50 V I C CNN
F3 "" 100 -50 50 V I C CNN
$FPLIST
 TO-???*
 D_*
$ENDFPLIST
DRAW
C 0 100 10 0 1 0 F
P 2 0 1 0 -100 -100 -100 0 N
P 2 0 1 0 100 -100 100 0 N
P 2 0 1 0 100 -100 100 0 N
P 4 0 1 0 -100 0 -100 100 100 100 100 0 N
P 4 0 1 8 -50 -100 -150 -100 -100 0 -50 -100 N
P 4 0 1 8 150 -100 50 -100 100 0 150 -100 N
P 4 0 1 8 150 -100 50 -100 100 0 150 -100 N
P 6 0 1 8 -75 25 -50 25 -50 0 -150 0 -150 -25 -125 -25 N
P 6 0 1 8 125 25 150 25 150 0 50 0 50 -25 75 -25 N
P 6 0 1 8 125 25 150 25 150 0 50 0 50 -25 75 -25 N
X A1 1 -100 -200 100 U 50 50 1 1 P
X K 2 0 200 100 D 50 50 1 1 P
X A2 3 100 -200 100 U 50 50 1 1 I
ENDDRAW
ENDDEF
#`

@jkriege2 If you like this maybe you could do your magic to spin out all the other versions? Sorry to add a perhaps burdensome request, but if you also see the benefits of this symbol then it makes sense to add them. And I'm also not sure the best way to name the symbols. If you don't have time, I'll probably get to it sometime.

jkriege2, 2017-02-08:
Hi!

hhmmmm ... About the different Version: I did them by Hand ...

About the Symbol itself: I would Limit it to 1 Version in the lib, but I'm not sure which one is better ...

@SchrodingersGat @diggit @pointhi What's your opinion on this?

JAN

SchrodingersGat, 2017-02-08:
I think if we are going to support "multiple styles" of the same thing then we should just make these diodes split-symbols and then the user can arrange them however they like on the schematic

evanshultz, 2017-02-09:
Separate symbols works for me. I'll be happy to update the existing symbols if that's the consensus.

While I can't find a good example online, consider current steering diodes around a FET to disuse the body diode (a Schottky pointed towards the drain and a FRED around that series pair with the cathode at the source). This is a common place to use a common anode diode package. Using these diodes, to me, it looks poor. This is another example where diodes placed as above or separate diode symbols is much more salient. Or consider the diodes here:
http://patentimages.storage.googleapis.com/US6731081B2/US06731081-20040504-D00000.png

Here's the symbol lib instance:
````
# D_Schottky_x2_KCom_AKA_Alt2
DEF D_Schottky_x2_KCom_AKA_Alt2 D 0 40 Y N 2 F N
F0 "D" 0 100 50 H V C CNN
F1 "D_Schottky_x2_KCom_AKA_Alt2" 0 -100 50 H V C CNN
F2 "" 100 -50 50 V I C CNN
F3 "" 100 -50 50 V I C CNN
$FPLIST
 TO-???*
 D_*
$ENDFPLIST
DRAW
P 2 0 1 0 50 0 -50 0 N
P 4 0 1 8 50 50 50 -50 -50 0 50 50 N
P 6 0 1 8 -75 25 -75 50 -50 50 -50 -50 -25 -50 -25 -25 N
X A2 1 150 0 100 L 50 50 1 1 P
X K 2 -150 0 100 R 50 50 1 1 P
X K 2 -150 0 100 R 50 50 2 1 P
X A2 3 150 0 100 L 50 50 2 1 P
ENDDRAW
ENDDEF
#
```

evanshultz, 2017-02-09:
Here's the current steering diode example I gave above in schematic form.

This:
[image]

Or this:
[image]

jkriege2, 2017-02-09:
Hhmmm ... I don't like the idea of separate diodes that much, because that is very error prone, because the common pin is not obvious and it is easy to connect the wrong pins in the schematic without noting it. take e.g. the component `MMBD4448HADW` from diode.lib:
[image]

If you don't take meticulous care it's easy to make something like that:
[image]
_"... it looked like 4 separate diodes ..."_

Best,
JAN

SchrodingersGat, 2017-02-09:
@jkriege2 makes a good point. 

I kind of prefer the symbol that @evanshultz suggests, at least for common cathode variant.

evanshultz, 2017-02-09:
Certainly I wouldn't want KiCad to be deceptive or prone to accidents. But personally, I think it's the responsibility of the schematic designer and don't see split symbols as a problem.

I can't think of any scenario where the current symbols communicate better the design intent. Having the diodes side by side looks better and fits easier into a schematic, to me. I'll make the updates if there a consensus to change.

Jan? Anybody else?

evanshultz, 2017-02-15:
And so, shall I convert the current dual symbols to ones with the diodes placed next to each other?

SchrodingersGat, 2017-02-16:
Yes, I think that will be the best outcome.

evanshultz, 2017-02-24:
I'm starting on this. Any preference on the default orientation? I'm thinking this:
- Common cathode and common anode point right
- Series point right at the top and lower one points left

Sound OK?

SchrodingersGat, 2017-02-24:
@evanshultz can you provide a quick diagram?

evanshultz, 2017-02-24:
Sure! Like this:
[image]

Let me know if this looks good and I'll knock out the other variations and submit a PR.

evanshultz, 2017-02-24:
@jkriege2 You always are good with the FPFilter. What would be possible for these symbols? I most often see them in SMT packages like SOT23, SOT323, etc. Could we do something like:
`D*`
`SOT*`
`*SC*`

jkriege2, 2017-02-24:
That's hard ... but since these are generic symbols, I would give no FPFilter at all (first guess) ... On the other hand for double-transistors we have these:
[image]

@SchrodingersGat @diggit @pointhi What do you think?

Best,
JAN

PS: I think such diodes are also available in e.g. TO-220, TO-247 ...

evanshultz, 2017-02-24:
Yes, you're absolutely right @jkriege2. They are in those TO packages.

I agree this is hard. They certainly wouldn't come in other 3 pin packages, like say an XLR connector. Is there a good method to segregate packages to any extent? The user could always turn on and off the filters as desired in CvPcb, but it would seem nice to show a reasonable (hopefully inclusive) selection of footprints by default instead of requiring the user to scroll past footprints that clearly aren't suitable. That, to me, would be the reason to provide a FPFilter for these symbols. Any ideas here?

And in addition to the FPFilter please comment on the look of the symbols above. Thanks!

poeschlr, 2018-01-20:
@evanshultz and @jkriege2 if this is still relevant please reopen it over at the new repo

closed 2018-01-20 by poeschlr