KiCad/kicad-library #1006: Leftover documentation without symbol/alias

diggit, 2017-02-15:
It looks like upstream KiCad have some new features about checking library files. After loading all schematic symbols, I get these errors:
```
04:20:26 PM: Alias 'TOP262YN' not found in library:

'/home/diggit/dev/kicad-library/library/powerint.dcm'

at line 1395 offset 14
04:20:26 PM: Alias 'AT90S4414-4PC' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 189 offset 19
04:20:26 PM: Alias 'AT90S4433-4AC' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 195 offset 19
04:20:26 PM: Alias 'AT90S4433-4PC' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 201 offset 19
04:20:26 PM: Alias 'ATMEGA164P-20PU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 447 offset 21
04:20:26 PM: Alias 'ATMEGA168PA-MM' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 573 offset 20
04:20:26 PM: Alias 'ATMEGA324P-20PU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 771 offset 21
04:20:26 PM: Alias 'ATMEGA48-20MMU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1011 offset 20
04:20:26 PM: Alias 'ATMEGA644P-20AU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1125 offset 21
04:20:26 PM: Alias 'ATMEGA644P-20MU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1131 offset 21
04:20:26 PM: Alias 'ATMEGA644P-20PU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1137 offset 21
04:20:26 PM: Alias 'ATTINY13-20MMU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1473 offset 20
04:20:26 PM: Alias 'ATTINY24A-MMU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1661 offset 19
04:20:26 PM: Alias 'ATTINY84-20SU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 1991 offset 19
04:20:26 PM: Alias 'ATTINY88-MMU' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 2129 offset 18
04:20:26 PM: Alias 'AVR-PUDI-6' not found in library:

'/home/diggit/dev/kicad-library/library/atmel.dcm'

at line 2678 offset 16
```

There is documentation in dcm file, but no symbol or alias have such name. They should be added or documentation should be removed.

evanshultz, 2017-02-15:
Thanks for catching this! I've submitted PR #1008 to address the powerint lib issue since I know that lib.

SchrodingersGat, 2017-02-24:
https://github.com/KiCad/kicad-library/pull/1013 dealt with the atmel files.

I think this issue is OK to close now - @diggit ?

diggit, 2017-02-24:
Those errors were fixed, but new appeared...
```
08:12:22 AM: Alias 'BDW93' not found in library:

'/home/diggit/dev/kicad-library/library/transistors.dcm'

at line 776 offset 12
08:12:22 AM: Alias 'IRF7607PBF' not found in library:

'/home/diggit/dev/kicad-library/library/transistors.dcm'

at line 2337 offset 17
08:12:22 AM: Alias 'IRF8721PBF-1' not found in library:

'/home/diggit/dev/kicad-library/library/transistors.dcm'

at line 2343 offset 19
```

Some versions of KiCad must be forgetting to delete documentation of aliases or sb. is editing files by hand a forgets same thing...

diggit, 2017-02-24:
This is weird, BDW93 looks ok to me :/

jkriege2, 2017-02-26:
This PR fixes the problems in transistors.lib (I think):
https://github.com/KiCad/kicad-library/pull/1062

diggit, 2017-02-26:
Everything seems to be fixed, thanks!

closed 2017-02-26 by diggit