KiCad/kicad-library #1012: some symbols broken in atmel.lib

pointhi, 2017-02-16:
When I add atmel.lib from the repository to my libraries, I get the following error messages when opening eeschema:

```
20:00:41: Alias 'AT90S4414-4PC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 189 offset 19
20:00:41: Alias 'AT90S4433-4AC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 195 offset 19
20:00:41: Alias 'AT90S4433-4PC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 201 offset 19
20:00:41: Alias 'ATMEGA164P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 447 offset 21
20:00:41: Alias 'ATMEGA168PA-MM' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 573 offset 20
20:00:41: Alias 'ATMEGA324P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 771 offset 21
20:00:41: Alias 'ATMEGA48-20MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1011 offset 20
20:00:41: Alias 'ATMEGA644P-20AU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1125 offset 21
20:00:41: Alias 'ATMEGA644P-20MU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1131 offset 21
20:00:41: Alias 'ATMEGA644P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1137 offset 21
20:00:41: Alias 'ATTINY13-20MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1473 offset 20
20:00:41: Alias 'ATTINY24A-MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1661 offset 19
20:00:41: Alias 'ATTINY84-20SU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1991 offset 19
20:00:41: Alias 'ATTINY88-MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 2129 offset 18
20:00:41: Alias 'AVR-PUDI-6' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 2678 offset 16
20:00:43: Alias 'AT90S4414-4PC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 189 offset 19
20:00:43: Alias 'AT90S4433-4AC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 195 offset 19
20:00:43: Alias 'AT90S4433-4PC' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 201 offset 19
20:00:43: Alias 'ATMEGA164P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 447 offset 21
20:00:43: Alias 'ATMEGA168PA-MM' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 573 offset 20
20:00:43: Alias 'ATMEGA324P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 771 offset 21
20:00:43: Alias 'ATMEGA48-20MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1011 offset 20
20:00:43: Alias 'ATMEGA644P-20AU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1125 offset 21
20:00:43: Alias 'ATMEGA644P-20MU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1131 offset 21
20:00:43: Alias 'ATMEGA644P-20PU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1137 offset 21
20:00:43: Alias 'ATTINY13-20MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1473 offset 20
20:00:43: Alias 'ATTINY24A-MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1661 offset 19
20:00:43: Alias 'ATTINY84-20SU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 1991 offset 19
20:00:43: Alias 'ATTINY88-MMU' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 2129 offset 18
20:00:43: Alias 'AVR-PUDI-6' not found in library:

'/home/thomas/Projekte/Kicad/kicad-library/library/atmel.dcm'

at line 2678 offset 16
```

SchrodingersGat, 2017-02-16:
It appears that these aliases exist in atmel.dcm but are not referenced anywhere in atmel.lib

Either:

1. We can work out which symbol (.lib file) each one is an alias of, and re-add them to the correct symbols
1. Delete the offending lines.

SchrodingersGat, 2017-02-16:
@pointhi I see you have suggested a fix - I will check that out now.

SchrodingersGat, 2017-02-16:
Thanks for raising this @pointhi

closed 2017-02-16 by SchrodingersGat

pwuertz, 2017-02-25:
Still seeing this after installing the current KiCad packages from the nightly PPA. What is the fix for this?

SchrodingersGat, 2017-02-25:
Do the nightly builds actually pull down the most recent libs? I think this
has been fixed (I've at least confirmed on my system).

Please download the Atmel lib from GitHub and see if that fixes the issue.

commit 130566db (2017-02-16): Fix #1012
commit 1b2aa9be (2017-02-16): Merge pull request #1013 from pointhi/atmel-fix-01

Fix #1012