KiCad/kicad-library #1026: Where to put shield footprints?

svheymann, 2017-02-20:
Is there a footprint lib for stuff like this:
[image]

or eg Arduino/Raspi shields?

Thanks!

jkriege2, 2017-02-20:
I would say there are two options:
1. You add a project template:
2. You add the actual (micro)controller board as a part that can be mounted on the lowe-side or upper-side of the PCB, see my current PR for Arduino: https://github.com/KiCad/Modules.pretty/pull/3

Best,
JAN

svheymann, 2017-02-21:
Hmm. What i try to have in the end is a symbol with multiple parts (for each pin header) like this:
[image]

Where all the signal names are already in place like this in eagle:

[image]

And when you add one of the connectors the footprint (or part of it, or you can delete the parts you dont need?) for this shield appears in the netlist/pcbnew.

I would find it more intuitive if a shield would be a "part" in the kicad-lib that you can simply add like a normal connector but with multiple sub-parts?

That would have the advantage if you like to add one of the connectorparts later in the design process you could do that using the lib symbol or am i wrong?

jkriege2, 2017-02-21:
Then, what I proposed in 2) is what you want to do ... whether the Symbol is split or not is a different Story (and I think Splitting should be OK here).

Best,
JAN

svheymann, 2017-02-21:
Do i understand that correctly, i should add the positive footprint (upper side of the board)?
So that someone can move the needed connectors (provided by the symbol) to the correct position on the lower side of the (negative) shield and then delete the board footprint?

jkriege2, 2017-02-21:
The footprint should contain all connectors ... If you want variants where only 2 of 5 connectors are present you should make several varying footprints with names describing that ... but I don't really see the sense in that ...

svheymann, 2017-02-22:
Ok. The PR i referenced adds the board with all connectors on which the shield can be mounted.

svheymann, 2017-03-22:
I splitted the 5 connectors to 5 different symbols in one lib. Now you can choose which connectors you use on the shield board and move it to the correct position using the board template in the PR. The symbols look like this:

[image]

In which lib would this fit in or should i create a new olimex lib?

jkriege2, 2017-04-26:
Maybe `modules.lib`... but I'm not sure about that ...

svheymann, 2017-05-02:
I put the new symbols into a PR.

closed 2017-10-06 by SchrodingersGat