KiCad/kicad-library #1032: Create a 4.0.6 tag

TheNikomo, 2017-02-22:
Right now 4.0.6-rc1 is shipping kicad-library tagged at 4.0.5, meaning it doesn't include any of the fixes that have been made to kicad-library since 4.0.5 release (my main gripe is PR https://github.com/KiCad/kicad-library/pull/917 currently).

Would it be possible to create a 4.0.6 tag for kicad-library so the rc builds can package that in the installer?
The tag should be updated to point to the final change for 4.0.6 when it is released, but I do not know what commit the tag should point at before that. I don't know of a way to make a tag always point at HEAD.

TheNikomo, 2017-02-22:
Talking on #git@freenode, trying to track HEAD with a tag would be a completely idiotic idea, and apparently using tags how KiCad is using them, is also stupid. So, that's not doable. If a tag is created, it'll have to point at a commit, or be manually rewritten once in a while.

nickoe, 2017-02-22:
We don't really want release branches on the libraries, this is no gain for us, mostly because we can not really resolve conflicts as easuly as can be done with typical source code.

TheNikomo, 2017-02-22:
Ideal solution for problem is probably to make RC builds pull in master instead of messing around with tags on kicad-library side

closed 2017-02-22 by TheNikomo

SchrodingersGat, 2017-02-22:
4.0.6 has been tagged