KiCad/kicad-library #1059: KLC inconsistence

Michi-F, 2017-02-24:
There is an inconsistence in the KLC itself:

> **3. General Rules for Symbols**
> 1. Using a 100mil grid, pin ends and origin must lie on grid nodes (IEC-60617).
> 2. For black-box symbols, pins have a length of 100mils. Large pin numbers can be accomodated by incrementing the length in steps of 50mils.

If you create a Pin with 150mils (according to 2. ok), its origin _and_ end can't lie an a 100mil grid (1. not ok).
I'd agree that the side of the pin where users make the connection should lie on the 100mil grid.

So either change 2. to "steps of 100mils" or remove "and origin" from 1.
I'd prefer the 2nd option, but I don't know if thats compliant with the IEC-60617.

What do you think about this ? Or do I misunderstand something ?

jkriege2, 2017-02-24:
see the new version of the KLC that we're currently working on!
https://github.com/SchrodingersGat/kicad-library/wiki/KLC

Michi-F, 2017-02-24:
OK ! Sorry, I didn't know that the KLC is being worked on there, and in the new Revision 1.2 it's fixed :)
As I expect the new revision will be here in the official repo soon, I close this issue.

closed 2017-02-24 by Michi-F

poeschlr, 2017-02-24:
Maybe this new version should be visible in some way. Such that people who make either footprints or symbols can see where the klc will head in the future and adjust to this fact.