KiCad/kicad-library #1068: Oscillators.3dshapes/Oscillator_SMD_SeikoEpson_SG8002CA-4pin_7.0x5.0mm.wrl model is missing

eldarkg, 2017-02-26:
Module that linked to this model:
https://github.com/KiCad/Oscillators.pretty/blob/master/Oscillator_SMD_SeikoEpson_SG8002CA-4pin_7.0x5.0mm.kicad_mod#L38

poeschlr, 2017-02-26:
Lots of 3d models are missing. The 3d shapes path in the 3d settings is set in all footprints because this way it is a lot easier for someone to add a 3d shape for any footprint.
So if you really want a 3d shape for it in the official repo, i would suggest you install freecad and the kicad-stepup plugin and design this 3d shape for yourself. (If you only want it for yourself you can search the internet for a 3d model and add it locally.)

Links:
freecad: https://www.freecadweb.org/wiki/Download
kicad stepup: https://sourceforge.net/projects/kicadstepup/

eldarkg, 2017-02-26:
ok

jkriege2, 2017-02-26:
Can be closed?

eldarkg, 2017-02-27:
@jkriege2 I think it is shouldn't be closed because it is actual

jkriege2, 2017-02-27:
It's not really an issue ... you can either provide a model, or not ... if not it's simply not there !

eldarkg, 2017-02-27:
Ok then I close it.

closed 2017-02-27 by eldarkg