KiCad/kicad-library #1083: pin header 3d shapes 1.27mm and 2mm

lostwithouthope, 2017-03-02:
hello,

wanted to say that the kicad project is awesome :)
I'm just wandering if there is a script for the generation of the pin header 3d packages and if it would be possible to extend for 1.27mm and 2mm  pitch pin headers.

cheers :)

poeschlr, 2017-03-02:
I fear that these models are generated with the old non step compatible scripts. We try to convert the 3d model library towards being more compatible with mechanical cad software. That's why we really would like to have 3d models in step file format.
The old 3d model creator lives under: http://cbernardo.github.io/kicad3Dmodels/

I would suggest you look into the scripts by @easyw. I think creating a script for the pin headers could be a good starting project for someone interested in this. These scripts create a 3d model in step and wrl format. (Which is what is preferred for new models.)
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator

easyw, 2017-03-03:
Hi @lostwithouthope , 
@SchrodingersGat Oliver already created the scripts to build pin-headers and box-headers for 2.54, 1.27, 2.00 pitch
I recently updated those scripts to match the generation of optimized 3d models both in STEP and WRL with material properties
requirements:
- FC 0.15 (best to create optimized STEP model)
- FreeCAD CadQuery 0.52 plugin

easyw, 2017-03-03:
fixed an issue on building models

lostwithouthope, 2017-03-03:
THX for the informations. @easyw does this mean you are going to open a pull request for those headers or should I take a look into it?

poeschlr, 2017-03-03:
@easyw  doesn't need to open a pull request. He is the owner of the repo. He can simply push to it. ;)
The latest commit seems to have fixed some problems with the box_header generator.

Shackmeister, 2017-06-14:
@poeschlr @lostwithouthope @easyw a small update on this is that I have made a lot of changes to these scripts meaning it can now generate angled connectors too. Ill push it as soon as Im done with fixing the footprints.. SMD models are also incoming :)  (btw I think the current status of the script is broken, ill add nu changes very soon)

Shackmeister, 2017-07-07:
In case you want to follow you can see the status of the new footprints and 3d models here:
https://github.com/KiCad/Pin_Headers.pretty/pull/24

Shackmeister, 2017-08-23:
All these models have been added to the official repo, Im closing this now :)

closed 2017-08-23 by Shackmeister