KiCad/kicad-library #1090: 74xx and 74xgxx library names are confusing

bradfa, 2017-03-07:
I'm interested in making a new 7400 series logic symbol, specifically the SN74LVC2T45 level shifter, but it's not clear to me from looking at the 74xx and 74xgxx libraries which one of those it should go into.  Additionally, looking at the 74xx and 74xgxx libraries, there doesn't seem to be any easily discernible rhyme or reason as to why parts are in one library and not the other.

For example, the SN74LVC1G332 part is in the 74xx library, but it clearly matches the 74xGxx expression which names the other library.

Most (all?) of the parts in the 74xgxx library are included in TI's "Little Logic" family brochures.  But there's some parts in the 74xx library which also fall into TI's "Little Logic" parts listings.

There are many of the templates using the 74xx library but no templates use the 74xgxx library.

Should there just be a single 7400 series logic library so as to avoid this confusion and so that people can easily find the 7400 series parts they are looking for?
Or is there a good reason why these should not be joined?

jkriege2, 2017-03-07:
THe logic libs are in dire need for a complete overhaul ... they e.g. still use hidden power-pins and massively violate KLC ... but currently it's not really clear how/who does that ... 

see also e.g.: 
- https://github.com/KiCad/kicad-library/issues/892
- https://github.com/KiCad/kicad-library/issues/605

So ... if you just want to add that level shifter ... just do as you feel right and try and make it as KLC-compliant as possible (see new KLC-draft: https://github.com/SchrodingersGat/kicad-library/wiki/KLC as it is nearly completed and will then replace the current KLC-version) ... I would not do larger changes before it is not decided what we do with these libs ...

JAN

bradfa, 2017-03-07:
@jkriege2 sorry, I guess my github issues searching was not very rigorous.  But searching again does show up quite a few other logic issues and the two you mention.

I would like to help cleaning this up so I will start participating on the other github issues which surround this topic, which now seems much larger than I had originally thought.

I am going to close this issue.  My concerns seem to be fully covered by #892, #605, and #1082.  No need for extra github issue noise.
Thanks! :)

closed 2017-03-07 by bradfa

keruseykaryu, 2017-03-07:
FYI: I was the creator of the `74xGxxx.lib`. It was entirely dedicated to the Single Gate series. 
IMO, if your level shifter is a single/double/triple gate, this library is right for him even if it has T in part name. If this is part of a standard 74xx series (not a part of Single Gate line) it should be in a standard `74xx.lib`.

bradfa, 2017-03-09:
On 03/07 09:27, Kerusey Karyu wrote:
> FYI: I was the creator of the `74xGxxx.lib`. It was entirely dedicated to the Single Gate series.
> IMO, if your level shifter is a single/double/triple gate, this library is right for him even if it [...]

OK.  The 74LVC2T45 (and the single version, too) has a few gates,
doesn't show up in the Wikipedia list of 7400 logic devices, but is
listed in TI's "little logic" family.  So I'm leaning towards it being a
part of the 74xgxx library.

Thanks for the info!  :)
-Andrew