KiCad/kicad-library #1091: 请问如何通过FREECAD软件输出3D模型？

Xduino, 2017-03-09:
你好！我自己用设计的后缀名为STEP的3D模型，用FREECAD输出WRL 2.0，可以在KICAD中显示，但是网上下载的一些STEP后缀文件，输出后无法显示。

jkriege2, 2017-03-09:
Could you ask your question in English please?

Best,
JAN

Xduino, 2017-03-09:
How do I output 3D model FREECAD and then import KiCad?
Hello there! I use the design of the suffix named STEP 3D model, with FREECAD output WRL 2.0, can be displayed in the KICAD, but the Internet to download some of the STEP suffix file, output can not be displayed.

Xduino, 2017-03-09:
@jkriege2 How do I output 3D model FREECAD and then import KiCad?
HI! I use the design of the suffix named STEP 3D model, with FREECAD output WRL 2.0, can be displayed in the KICAD, but the Internet to download some of the STEP suffix file, output can not be displayed. Thank U！

jkriege2, 2017-03-09:
@easyw Could you comment on that?

easyw, 2017-03-09:
@Xduino 
the easiest way to do it is through [KiCad StepUp](https://sourceforge.net/projects/kicadstepup/)
you can follow a [howto video](https://youtu.be/O6vr8QFnYGw)
and get some tips at KiCad forum
[how-to-interactive-align-3d-model-to-pcbnew-footprint-new-tool-for-kicad-in-altium-style](https://forum.kicad.info/t/how-to-interactive-align-3d-model-to-pcbnew-footprint-new-tool-for-kicad-in-altium-style/1711)
Recently many 3D STEP models have been merged in to the [official 3D MCAD repo](https://github.com/KiCad/packages3D)
If you need further assistance you can find me  KiCad Forum
Cheers,
Maurice

Xduino, 2017-03-13:
@easyw Hi！ Will "kicad StepUp tool" how to install the freecad toolbar above it? thanks！
[image]

easyw, 2017-03-13:
as in the kicadStepUp-starter-Guide.pdf @ pg7
http://www.freecadweb.org/wiki/Customize_Toolbars

Shackmeister, 2017-07-01:
@Xduino Hello, in case you got the answers you needed can you please consider closing this issue?

Shackmeister, 2017-07-07:
@Xduino hi, ill close this now, if you want you can reopen it

closed 2017-07-07 by Shackmeister