KiCad/kicad-library #1092: Problem with loading of transistors library (part: BF240)

www2000, 2017-03-09:
I want to report an problem with loading of the transistors library that i download from the ubuntu ppa (version 0.1-0 201609051216).

Error/warning:
Alias 'BF240' not found in library:

'/usr/share/kicad/library/transistors.dcm'

at line 830 offset 11

SchrodingersGat, 2017-03-10:
Thanks @www2000  - fixed here - https://github.com/KiCad/kicad-library/pull/1094

SchrodingersGat, 2017-03-14:
@www2000 can you please confirm that the update has fixed your problem? :)

www2000, 2017-03-14:
I forgot close the issues.
And i can confirm that you fix the problem

closed 2017-03-14 by www2000

kharar, 2017-12-07:
I got that problem on Version: (2017-11-24 revision a01d81e4b)-makepkg, release build

poeschlr, 2017-12-07:
@kharar your version seems to be a nightly build. Nightly builds (at least for windows and ubuntu) include a snapshot of the lib of the date they are build.
I looked at the library from that date and found that since 5. oct the lib is named Transistor.lib. That version should include the fix. (could not find any mention of the BF240 in the dcm file.)

This leaves me to believe that you did not update the libs with your kicad version. I would suggest you manually download the lib tagged as v4.0.7 from the [release page](https://github.com/KiCad/kicad-library/releases).

poeschlr, 2017-12-07:
Ok while i wrote my response it seems you also commented on a  different issue. So i am no longer certain what your problem is. Could you give the detailed error message? (And tell us what lib is affected?)

If your error message is indeed the same as reported by @www2000 then my post above applies.
If it is the same as reported by @pointhi in the other issue, then you might not yet have the fix that was made in response for that particular issue. (Your nightly was released quite near the date where that issue was reported.)

The safest bet would still be to use the stable library instead of relying on the lib that is shipped with nightly.

kharar, 2017-12-22:
Hi, sorry for the confusion, I had the impression that this thread was not BF240 specific. If I am wrong in my assumption you may disregard of my previous messages.
My problem came after upgrading to the nightly and was related to custom libraries that I made myself using the 4.0.7.
Since then a lot have happened on my computer and I am no longer sure that I will be able to recreate this specific error.