KiCad/kicad-library #1096: Error when starting CvtPCB

sidprice, 2017-03-10:
Errors were encountered loading footprints:
PARSE_ERROR: Expecting 'circle, rectangle, oval, or trapezoid' in input/source 'https://github.com/KiCad/Connectors.pretty' line 34 offset 14 from C:/Jenkins/workspace/windows-kicad-msys2-stable/src/kicad/common/dsnlexer.cpp : Expecting() : line 369
Sid

poeschlr, 2017-03-10:
We already know about it.
Someone added a footprint with a rounded rectangle to the library. Sorry about that. (Hopefully one of the maintainers will do something about this.)
https://github.com/KiCad/Connectors.pretty/issues/66

jkriege2, 2017-03-11:
fix is merged: https://github.com/KiCad/Connectors.pretty/pull/68

sidprice, 2017-03-11:
Excellent ... thanks.

closed 2017-03-11 by sidprice