KiCad/kicad-library #1100: Plan for naming convention

Gasman2014, 2017-03-12:
Is there an ultimate plan for the final naming convention for the modules? This does not seem to be discussed in the KLC. 

I notice that there have been some sensible name changes made to various components e.g. Resistors, Capacitors and Inductors to xx_SMD.shapes & xx_THT.3dshapes - but this does not yet apply to the Diodes - which are still Diodes _SMD.3dshapes and Diodes_ThroughHole.3dshapes. Similarly for Chokes which are Chokes_SMD or Chokes_Toroid_ThroughHole.3dshapes.

There is some further inconsistency in the naming of some of the other libraries too - especially with regard to use of manufacturer names. e.g. Transformers and NF_Transformers_ETAL

An agreed final naming convention target would be helpful as part of the KLC. I think quite a few people find it difficult to find the appropriate footprint in the current naming structure. Perhaps a tree structure of the ultimate goal would be helpful?

poeschlr, 2017-03-12:
Look at the current klc draft.
https://github.com/SchrodingersGat/kicad-library/wiki/KLC
Especially the first paragraph
And the first paragraphs of the symbol and footprint sections.

What would you add to make it more clear.

jkriege2, 2017-03-12:
Hi!

a few comments:
1. I think `_ThroughHole` will be removed eventually. The problem is, we cannot simply delete a repository, as KiCAD will crash when it doesn't find a repo explicitly listed in a project-file (as far as I understood that)
2. nevertheless we're in the process of reorganizing and the effort is quite far by now. Note please that several Repos have been labeled deprecated (see https://github.com/KiCad/kicad-library/blob/master/template/fp-lib-table.for-github for a list, just search for "deprecated"). Also I hope that we put that note into the Readme.md displayed when opening any Repo on GitHub, see e.g. https://github.com/KiCad/Air_Coils_SML_NEOSID.pretty 
3. That said: The coild have been reorganized into 4 new repos:
  - https://github.com/KiCad/Inductors_SMD.pretty
  - https://github.com/KiCad/Inductors_THT.pretty
  - https://github.com/KiCad/Transformers_SMD.pretty
  - https://github.com/KiCad/Transformers_THT.pretty
In that process all old symbols were merged into the new repos and partly renamed in the process!
4. We're slowly converting the 3D repos to that naming convention too ... but that takes time, as it also requires changing the links in the footprints!

If there's anything left after these 3 points, please let us know, so we can fix it ;-)

Best,
JAN

Gasman2014, 2017-03-12:
Thank you both, Rene & Jan. That is very comprehensive. The link that Rene provides is particularly helpful and the workflow explanation from Jan is very clear.

I would point out that the link to the KLC on this Github repo takes you to a page which is less comprehensive (https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention) which is why I raised the issue. Perhaps the new, proposed KLC summary could be linked as an additional document to this repo?

jkriege2, 2017-03-12:
the new KLCs are currently still discussed (although mostly complete) and should go live (i.e. replace the old version) in the next days (hopefully). You can follow the discussion here: https://github.com/KiCad/kicad-library/issues/879

Can this issues be closed then?

Gasman2014, 2017-03-12:
Yes, please. My only comment is that the final KLC roadmap might have been made clearer ahead of the start of implementation but understand that things evolve! I appreciate that the current implementation is somewhat 'sub-optimal' ;) and that further changes will devolve from the move to .sweet / .richio .

closed 2017-03-12 by Gasman2014