KiCad/kicad-library #1109: Which soft is recommended for creating/editing 3d models?

joe-skb7, 2017-03-19:
I want to modify existing 3D model (QSOP-16) to create new one (QSOP-24). Could someone advice, which software should I use (Wings 3D, FreeCAD, etc)? Ideally I'd like to use the same software that was used to create original model (QSOP-16), if it's available for Linux.

Also, it would be great if you can give me a few hints how to start with that modeling software. I mean, some tutorial links, maybe even something KiCAD related.

Thanks.

SchrodingersGat, 2017-03-19:
FreeCAD would be the one to choose as it is used to make the KiCad models (in both STEP and WRL format).

A lot of the KiCad 3D models are scripted using FreeCAD - https://github.com/easyw/kicad-3d-models-in-freecad

For a good source of new models look here - http://github.com/kicad/packages3d

@easyw @poeschlr @jkriege2 are much more skilled in FreeCAD than I am so they may be able to offer extra help.

joe-skb7, 2017-03-19:
@SchrodingersGat Thanks, Oliver. Good info for starters.

@michal777 Hi Michal. I see that QSOP 3D models were added by you. Do I understand correctly that you are the author? If so, can you advice the best way for me to create QSOP-24 (probably using QSOP-16 as a template)? I mean, which soft is best suited for this, which package dimensions to use, maybe recommended workflow, etc.

Thank you.

poeschlr, 2017-03-19:
There are scripted models for lots of ic packages already. You might want to look here:
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT
The ssop or tssop scripts should be easily adaptable to create qsop models.

poeschlr, 2017-03-19:
If you create the model in any other way than by scripting them as i suggested above, don't forget to add the source file to: https://github.com/poeschlr/packages3D-source

The 3d files should also go to the new repo: https://github.com/KiCad/packages3D
Make sure you supply a step model and a correctly scaled wrl model.

When using freecad to design the model, use kicad stepup to export the step and wrl files.
https://sourceforge.net/projects/kicadstepup/

Use freecad 0.15 to export the step file. There is a major bug in freecad 0.16 and in 0.17 which creates unusable step files. (It adds an assembly to the file. Some software packages can not use such a file.)

SchrodingersGat, 2017-03-19:
@poeschlr did you mean to link to your copy of packages3d-source (rather than the KiCad one)?

poeschlr, 2017-03-19:
Sorry. fixed the link.

easyw, 2017-03-20:
@joe-skb7 if you are looking for the recommended way to create a QSOP-24 3D model I really would suggest you to use the scripted way....
You just have to add your dimension parameters in the parameter file of SSOP folder and you will have your model...
https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/GW_QFP_SOIC_SSOP_TSSOP_SOT
http://www.onsemi.com/pub/Collateral/492B-01.PDF
requirements:
FC 0.15 and CadQuery 0.52

SchrodingersGat, 2017-03-20:
@joe-skb7 I would second the suggestion of @easyw above. If you add the parameters to his script, then we can add the resulting model to the official 3D repo! - http://github.com/kicad/packages3d

joe-skb7, 2017-03-20:
@poeschlr @easyw Thank you for your input. I've tried to build FreeCAD 0.15, as suggested (as I didn't find FreeCAD 0.15 package for Debian testing, which I'm using), but the build fails (something related to <Eigen/LeastSquares> include).

Do you guys know what exactly wrong with FreeCAD 0.16? Can I just revert some commit to get rid of that bug (which adds assembly to step files)? If you know any details on that issue, please share. It's really inconvenient, probably it's good idea to ask FreeCAD developers to fix that for us.

Anyway, right now I feel like I'm dead in the water. So any suggestions on how to proceed further are really appreciated.

Thanks.

easyw, 2017-03-20:
I don't have Debian, I can test it only on ubuntu ...
Do you have CadQuery installed and working?
Maurice

poeschlr, 2017-03-20:
To be honest i currently use freecad 0.15 in windows for exporting models.
The bug is known by the developers and @easyw told them that their workaround does not work.
See bug report:
https://forum.freecadweb.org/viewtopic.php?f=3&t=20455

For your python error: make sure that freecad is in the python path.

easyw, 2017-03-20:
@joe-skb7 FC 0.16 should work, just the STEP model generated will have surplus PRODUCTS ...
if you digit the following commands in 'python console' of FC:
```
import FreeCAD, FreeCADGui
import ImportGui
import FreeCADGui as Gui
from Gui.Command import *
```
what do you have in 'report view' of FC?

joe-skb7, 2017-03-20:
@easyw After executing those commands, report view is empty, but python console has:

>Traceback (most recent call last):
>  File "<input>", line 1, in <module>
>ImportError: No module named Gui.Command

easyw, 2017-03-20:
very weird... 
Could you please try to edit the file **main_generator.py**
and comment the line 79
```
# from Gui.Command import *   commented for Debian
```
then save and retry the building just typing:
`$ freecad ./main_generator.py`
Thx
Maurice

joe-skb7, 2017-03-20:
@easyw Commenting out `from Gui.Command ...` only `main_generator.py` doesn't help. So I also commented out the same line in `_tools/cq_cad_tools.py`. After that mentioned error disappears, but I see this error:

> missing CadQuery 0.3.0 or later Module!
> https://github.com/jmwright/cadquery-freecad-module/wiki

and this one:

> Exception while processing file: ./main_generator.py [name 'cq' is not defined]

joe-skb7, 2017-03-20:
@easyw Ok, I installed CadQuery in wrong place (`/usr/local/lib/freecad/Mod`). Now I extracted it to `~/.FreeCAD/Mod` and FreeCAD finally sees it. But some other errors occur... I will add update here shortly.

joe-skb7, 2017-03-20:
@easyw I uncommented `from Gui.Command ...` lines (that I commented before), and now it seems to be working. So the actual error was the wrong installation path for CadQuery. One should use `~/FreeCAD/Mod`, as stated at https://github.com/jmwright/cadquery-freecad-module/wiki.

 Thanks a lot for your help :) I'll try to send QSOP-24 package shortly.

easyw, 2017-03-20:
:thumbsup:

closed 2017-03-20 by joe-skb7