KiCad/kicad-library #1123: Body diode backwards on BSS138 et al.

horvathp, 2017-03-25:
I believe the body diode is drawn with opposite polarity on the transistors.lib/BSS138 (2N7002 etc.) model. 
https://www.fairchildsemi.com/datasheets/BS/BSS138.pdf
On a side note, there are generic MOS/NPN/PNP etc. transistors in device.lib with different pin arrangements. I think it would make sense having generic NMOS/PMOS with body diode in device.lib as well.

horvathp, 2017-03-25:
Furthermore, BSS138 has aliases
TP0610T and VP0610T.
However, these are P-Channel, is BSS138 N-Channel. 
Isn't this a mistake?

cpavlina, 2017-03-26:
Yeah, you're correct, it's backward, and the aliases are a mistake.

SchrodingersGat, 2017-03-27:
@horvathp thanks for pointing this out - could you perhaps submit a PR fixing this issue?

jkriege2, 2017-03-27:
@horvathp a side-note: The NMOS/PMOS in device.lib do HAVE a body diode ... a fix of transistors.lib would be great!

horvathp, 2017-03-31:
Indeed they have, I was looking at an older version.

SchrodingersGat, 2017-03-31:
Fixed in https://github.com/KiCad/kicad-library/pull/1135

closed 2017-03-31 by SchrodingersGat