KiCad/kicad-library #1124: KLC Convention re: Reference Designator

kaitrek, 2017-03-27:
The reference designator for footprints in the current KLC convention is stated to be "placed on the fabrication layer."
That does not make sense, since people always prefer to see the ref. designator on the silkscreen instead. If the rule is there so that the reference designator is useful for assembly houses, then the rule needs to be amended to say that the reference designator needs to be place on silkscreen and fabrication layer on top of each other.

Furthermore, if it has to go on both layers, then kicad developers need to be alerted so they could add a 3rd box in the Footprint Editor Options for "Default values on new footprint creation:" so that REF** could be added to Silkscreen and Fabrication layers by default.

SchrodingersGat, 2017-03-27:
Hi @kaitrek there is a new KLC underway that will be made official soon, and this issue is addressed there: 

https://github.com/SchrodingersGat/kicad-library/wiki/KLC#7-general-rules-for-footprints

SchrodingersGat, 2017-05-04:
@kaitrek ok to close this issue?

kaitrek, 2017-05-04:
@SchrodingersGat yes, thx

closed 2017-05-04 by kaitrek