KiCad/kicad-library #1131: Incompatible requirements for symbol conventions?

SteveevetS, 2017-03-30:
KiCAD Library Convention
3. General Rules for Symbols
1) Using a 100mil grid, pin ends and origin must lie on grid nodes (IEC-60617).
2) For black-box symbols, pins have a length of 100mils. Large pin numbers can be accomodated by incrementing the length in steps of 50mils.
3) Origin is placed in the middle of symbol.

It is not possible to always meet "1)" and "3)" simultaneously while maintaining symmetry and minimum size for a symbol.  Consider a simple 8 pin symbol, with 4 pins arranged on each vertical edge of the symbol outline.  In this case it is necessary to use the 50mil grid to maintain symmetry, unless one is willing to leave a blank pin position in the middle of each set of four pins.  Note that a symbol drawn on the 50mil grid will be consistent with the default 50mil grid recommendation for EEschema, and pin ends can still be multiples of 100mils apart.  The symbol origin will still be at the origin of the editor.  To fix this issue, I suggest that "1)" read as follows:
1) There must exist a 100mil grid, such that all pin ends lie on grid nodes (IEC-60617).  The symbol may be shifted by 50 mils in any direction to line up with said 100mil grid in the Part Library Editor.

SchrodingersGat, 2017-03-30:
@SteveevetS the [new KLC version](https://github.com/SchrodingersGat/kicad-library/wiki/KLC) is almost out - and this wording has been cleaned up there.

SteveevetS, 2017-03-30:
Great - Thanks!

SteveevetS, 2017-04-03:
I think the issue still remains.  Please see the items with an "*" below from the newly released KCL, and my original post above.  In practice, symbols are being drawn with a 50mil grid with their origins on the 50mil grid,  even though the pin origins are on "a" (not "the") 100mil grid.  This was personally a source of confusion for me, so I think others will find the same issue.  I think it should be clear that the 50mil grid must be used for symbols, with pin origins on _**A**_ 100mil grid, not necessarily _the_ 100mil grid.

4. General Rules for Symbols
    Pin placement
        *Using a 100mil grid, pin origin must lie on grid nodes (IEC-60617)
        Pins should have a length of at least 100mils (2.54mm)
        Pin length should not be more than 300mils (7.62mm)
        Pin length can be incremented in steps of 50mils (1.27mm) if required e.g. for long pin numbers
        Shorter pins may be allowed for simple symbols such as resistors, capacitors, diodes, etc
    Symbol visual style
        *Origin is placed in the middle of symbol

evanshultz, 2017-04-03:
@SteveevetS Can you explain the issue in more detail? If the pins are an even number, and spaced equally, then the symbol will have empty area at the top or bottom. But that doesn't violate KLC in any way. The origin can still be in the middle with all pins on a 100mil grid.

50mil grid isn't required for symbols. Only pin origin (end) must be on a 100mil grid. Otherwise, there are no requirements for the placement location of elements.

I believe I see what you're saying, and it is true that many symbols in the existing libraries are not KLC-compliant, but I don't see why "the" or "a" has any bearing on the KLC requirements. Certainly there is a barrier to entry for any sufficiently complex set of rules, like KLC, but I would like your help to better understand your perspective since I'm not seeing it.

hackscribble, 2017-04-03:
In this example, pins are on the 100mil grid and the symbol centre is at (0,0).  The symbol is taller than it needs to be.

[image]

In this example, pins are on the 100mil grid but the symbol centre is not at (0,0).  Looks better but breaks KLC.

[image]

Finally, here the pins and symbol are on 50mil grid, the symbol centre is at (0,0) and the pins are spaced 100mil apart.  I think this is what @SteveevetS is suggesting, and I agree.  But it needs a change to the KLC.

[image]

jkriege2, 2017-04-03:
Please note that there can also be exceptions to certain rules of the KLC, if there is a reason. Usually the center-rule is seen as less important than the 100mil grid. So usually we place the center a bot off from the symbol center to get the pins on THE 100mil grid.

Best,
JAN

hackscribble, 2017-04-03:
Thanks, @jkriege2 

So the second of the three examples is the preferred style.

jkriege2, 2017-04-03:
I would say that's what we've done most of the time ...

evanshultz, 2017-04-03:
It's situation-dependent. The first example is fine too, IMO. Perhaps the pins are spaced apart at +100mil and -100mil above and below the x-axis. Or if there are 3 pins on the side, then all is balanced. I've seen a number of styles in the library that all meet KLC and are acceptable?

Has this question been answered?

SteveevetS, 2017-04-03:
Hackscribble hit the nail on the head.  I much prefer the third figure for its symmetry, and I've seen examples of this in the Libraries.  This does violate KLC, though, as written (or at least it's unclear), so this is why I'm advocating a change or clarification in KLC.  The pin origins of the 3rd symbol are not on **_THE_** 100mil grid, but they are on _**A**_ 100mil grid.  Is there any disadvantage to using a 50mil grid for symbols with the added constraint that pin-origins must be on "A" 100mil grid?  

In my library, I have even less constraint.  I allow pin origins to be on a 50mil grid (no 100mil grid exists to coincide with all pin-origins), but I don't allow any two pin origins to be less than 100mils apart.  This results in the smallest possible, symmetrical symbol, and it works fine as long as you use a 50mil grid in Eeschema (the default and recommended grid).  I think it also achieves the intent of "100mil grid" for pin-origins.  In the example attached , only pins 4, 1, and 8 are on THE 100mil grid.  Is there any reason that this symbol would be objectionable?

[image]

bobc, 2017-04-04:
It depends how you interpret "middle" in "Origin is placed in the middle of symbol". To engineers, middle can mean the exact center to 1 micron, but they might also ask "centroid or circumcenter?". In layman's use, "middle" just means somewhere near the center.

We just need a middle point which is somewhere around the visual center of the symbol, the origin has no electrical, mechanical or other technical significance. It's purely for aesthetic and consistency reasons.

If you like, the rule could be clarified -

    Origin is placed in the middle of symbol (to nearest grid node)

And there is no incompatibility or functional change.

SteveevetS, 2017-04-04:
Yes, I (an engineer) agree that indicating that "middle" is just to the nearest 100mil grid node would clarify the rule.  Is the intent to allow symbols to work with a 100mil Eeschema grid?  If so, then I'm satisfied that simply clarifying the definition of "middle" will suffice.  The reason that I ask the question is that the default and recommended grid for Eeschema is 50mils.  I'm not advocating that pins have 50mil spacing, but asking that symbols constructed on a 50mil grid with 100mil pin spacing be considered. - Thanks.

Misca1234, 2017-10-22:
> Using a 100mil grid, pin ends and origin must lie on grid nodes (IEC-60617).

I find it amazing that KiCad community even thinking of adhering to to this rule, well if the rule is even in there.
This mil (milli-inch) thing is the worst and most stupid "invention" within the engineering community since the introduction of the metric system.

Now, there was time when metric and imperial system was competing a little (a couple hundreds years ago, 1 or 2 month right after the French revolution) but that battle was over long time ago. 
We all also find the story about the standard yard in wood that burned up a little amusing but today none is using imperial/USCS/English units except for some people in backwater communities or children that suffer from an upbringing under harsh conditions by parents that dreamed of the good old days when youth knew their place.

Since the last 50 years or more the yard (and therefore the inch) is defined as a scale factor of the meter, so the relation ship between meter and inch is not even an irrational number any more, just a scale factor, which diminish the need for inch even less.

Now, those of us that rememberer TTL logic and the DIL capsules understand why and where this mil crap comes from but it does not automatically justify the usage of this and without any doubt it shows that the stubborn old backwater engineers are active within the IEC community.

Even countries that used to use imperial/USCS/English units don't even use the term mil, because they get an internal mess in the documentation them-self because its similarity to the short from of millimetres, they use the word "thou".

So justify or argue for the usage of mils by pointing on IEC-60617 means that the same goes for all other rules stated in IEC-60617. It also means that any work within KiCad community which touch an IEC XXXX must follow that IEC XXXX.

This means that either the mil crap is accepted which also means that any work, action or contribution that violates any IEC XXXX can not be accepted.

But if KiCad community accepts any deviations from any IEC XXXX it means that KiCad community does not need to adhere to the mil crap in IEC-60617.

poeschlr, 2017-10-22:
We only need this 100mil stuff until eeschema can snap wires to pins. Currently if a pin is not on grid you can not connect to it. 
100mil is the highest grid in eeschema. (Or most coarse grid? No clue how to communicate this in english.) This means if the symbol has it's pins on this grid then it works on any grid that is available in eeschema. 
Another possibility would be the 50mil grid. (next step down.) We allow this for small symbols. (Of course this means this symbols can not be used with eeschema set to 100mil grid.)

bobc, 2017-10-22:
AFAIK from an old copy of IEC-60617 is that it says "All of the symbols are designed so that their connection points fall on a modular grid" which is eminently sensible. It does not seem to prescribe a specific grid size.

As @poeschlr describes, 100 mil grid is chosen for practical reasons. We could I suppose specify a 2.54mm grid in the KLC, would that help?

Either way, for Sunday morning relaxation I recommend Frank Sinatra on the music player, jam on toast and a nice cup of tea :) There are lots of things about the modern world to stress about, but a legacy use of imperial measurements is not one of them...

jkriege2, 2017-10-22:
+10 for the jam&tea ;-)

SchrodingersGat, 2017-11-21:
@SteveevetS please refer to the new KLC at http://kicad-pcb.org/libraries/klc/

If you feel that this issue has been resolved to your liking, please close this issue. Thanks :)

poeschlr, 2018-01-20:
I assume this has been addressed here and in the new KLC. If not please reopen this over at the new repo https://github.com/KiCad/kicad-symbols

closed 2018-01-20 by poeschlr