KiCad/kicad-library #1132: Suggestion to add "%R" to Fabrication layer inside each component outline (in addition to Silkscreen layer; where it is NOT inside each component outline).

SteveevetS, 2017-03-30:
On dense boards with 0402 or smaller components, The Silkscreen layer makes for a very poor assembly drawing.  It is much more readable to have an assembly drawing with each component's reference designator actually inside the Fabrication layer outline of that particular component , scaled in size to fit the component.  I suggest that the KLC contain provision for including "%R" inside each Fabrication outline (or projection of the Courtyard outline on the Fabrication layer).  I sugest that the font size of "%R" be the largest font that will fit within the given Fabrication layer rectangle for a 5-character reference designator.

SchrodingersGat, 2017-03-30:
Great suggestion! And one that's in the [new KLC](https://github.com/SchrodingersGat/kicad-library/wiki/KLC) :)

SteveevetS, 2017-03-30:
Excellent!  I'll check it out - thanks!

closed 2017-03-30 by SteveevetS