KiCad/kicad-library #1133: TO_SOT_Packages_SMD:SOT-23_Handsoldering 3D Placement

kaitrek, 2017-03-31:
The 3D placement settings needs to be changed to:
+90 degrees Z axis
-0.05 Z offset

Otherwise the object is floating and rotated by 90 degrees.

SchrodingersGat, 2017-03-31:
@easyw @jkriege2 has this been fixed with the recent updates to the 3D models in the packages3d repo? Perhaps those models need to be shifted here.

poeschlr, 2017-03-31:
It seems the _Handsoldering files are not script generated. (At least not in the one that lives in the packages3d repo.) I think the footprint should be updated in such a way that it uses the same 3d model for the _Handsoldering variant as for the normal variant. (As stated in the new KLC)

evanshultz, 2017-03-31:
I checked all the SOT* footprints at https://github.com/KiCad/TO_SOT_Packages_SMD.pretty and found the following that were not placed with 0 location, 0 rotation, and 1:1 scale:
- PQFN_8x8
- PowerMacro*
- SOT-404
- SOT-428*
- SOT-555
- SOT-665
- SOT-666
- SOT-669*
- SOT-89
- TO-252
- TO-263
- TO-277
- TO-50

I'll submit a PR to correct those footprints unless somebody objects.

SchrodingersGat, 2017-03-31:
@evanshultz  please do that - the origin / scaling / rotation should be normalized.

evanshultz, 2017-04-01:
Done. PR is at https://github.com/KiCad/TO_SOT_Packages_SMD.pretty/pull/41.

Shackmeister, 2017-06-12:
@SchrodingersGat @kaitrek  hi guys  this has been fixed when I pushed all the new SOT models so you could consider closing this issue

SchrodingersGat, 2017-06-13:
@kaitrek are you ok with closing this PR?

kaitrek, 2017-06-13:
Can't look at the commits now, assuming other people have checked. Closing.

closed 2017-06-13 by kaitrek